dresden-elektronik/deconz-rest-plugin #2896: Phoscon and Deconz issues

meremortals70, 2020-06-08:
Hi there 

On a couple of recently added devices to Deconz having some issues.

**First Issue**

When the SZ-ESW01 plugs are added to Deconz, they have different names.  This normally isn't an issue as when you change the name in Phoscon it updates in Deconz.  However, when these plugs which are detected as lights, are renamed in Phoscon the names are either not updated in Deconz or are updated but not persistent so when Deconz is restarted the names revert.

[image]

Second Issue

When the SZ-DWS04 are added the don't appear in Phoscon however will appear under the Deconz Integration in HA,  they don't appear in Phoscon.  They cannot be renamed here which normally propagates up to HA and down to Deconz.  When renamed in the HA Deconz integration this does not propagate down to Deconz. when Deconz is restarted they reset their names and as a consequence become unavailable in HA as I have renamed them there.

[image]

This is the same for some Tuya Temperature/Humidity Sensors.

[image]

I have downgraded deconz to .75 due to the issues with .76 and .77

Mimiix, 2020-06-08:
Hi!

Can you please move the issues regarding to Phoscon to this repository: https://github.com/dresden-elektronik/phoscon-app-beta

I will create a PSA later today and making clear to post stuff at the correct places, as this hasnt been communicated before.

On the other questions, i cannot answer. However, i've noticed that more people have issues with devices not keeping their renamed name.

I will bring that up to @manup later today.

Thanks!

meremortals70, 2020-06-08:
Well the issue is I don’t know if it is a Deconz or a Phoscon issue as it’s about the interaction between the two

Mimiix, 2020-06-08:
Devices not showing up in Phoscon is something that is phoscon related. Devices need to be added there manually by @YKO-de .

Mimiix, 2020-06-08:
You can keep it open for the remaining issue on renaming devices :)

zombielinux, 2020-07-08:
This is still an ongoing issue. I have several of the Tuya Temperature and Humidity sensors, all coming in a RH3052 (n) where n is some integer. I cannot rename them at all. 

This is with the latest docker image as well.

SwoopX, 2020-07-08:
Rename them via REST API.

zombielinux, 2020-07-08:
Confirmed the REST api works. Still broken in the deConz application though.

ebaauw, 2020-07-08:
There's no function in the GUI to rename a node.  You can set the _User Descriptor_ from the _Node Info_ panel, but I have yet to see a device that supports this descriptor.  The name on the node is set by the REST API plugin when the name of (one of the) corresponding resource(s) is changed through the API.

manup, 2020-07-09:
Indeed the User Descriptor is not used much anymore, I also only know the FLS-PP lp which supports it. In the Zigbee specification it's an optional feature and presence of the User Descriptor is visible in the Node Descriptor flag.

Mimiix, 2020-07-09:
@manup @ebaauw So this is a wontfix?

closed 2020-08-08 by stale[bot]

kyletpugh, 2021-01-27:
Has this been fixed? Because I am having an issue with names being saved (using beta channel packages). I have tried from the REST API, the phoscon app, and the old web interface. The names are always wrong when deCONZ is restarted, I have to go through and rename about fifteen devices every time.

Smanar, 2021-01-27:
Perhaps here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4255#issuecomment-768090804
If name are still good in phoscan but bad in deconz.