dresden-elektronik/deconz-rest-plugin #2897: PSA: Issue posting guidelines

Mimiix, 2020-06-08:
Hi there!

As you might have noticed, i came in with a broom and cleaned up some old issues. However, i might have closed stuff that hasn't been fixed. I'm sorry if that happened :( 

From now, i'd like to add some guidance in how we should post items here. I'd like to ask all of you to please follow the following guidance rules in order to keep this place tidy 😄:

- Decide where your issue suits best. Post in this repository if it regards deCONZ. Post in this one if it regards Phoscon: https://github.com/dresden-elektronik/phoscon-app-beta

- If your post is about a device request, please start the title with [DEVICE REQUEST] and add information as described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

- If your issue is solved or either way ended, please close it yourself ;) I really try to stop going back into these long issues with question after question. It makes things messy.

- Please search in the closed + open issues if your request/question/problem is posted before. Read from there, it might be something that already has been answered. If not, please open a new issue :)

- Stay civil with each other. Being negative doesn't help things. We are a community and should do this together :)

- Do not post any URL's to "other" PR's or compiled DE plugins you created yourself outside of this repository, as we are not able to verify it's authenticity. You are free to create a PR in this Repository. 

- If you have any quick questions, reside to [Discord](https://discord.gg/vj7QAGM). That way you probably get a answer way quicker. 

If anything comes up, i'll be updating this.

Kind regards,
Dennis

Hypfer, 2020-06-09:
I'd suggest using issue templates where you can both automatically attach labels as well as give guidance inside each new issue form.

You can use comments there which will not appear in the final issue as well as have types which direct you elsewhere.

See the [home assistant core repo](https://github.com/home-assistant/core) as an example

Mimiix, 2020-06-09:
> I'd suggest using issue templates where you can both automatically attach labels as well as give guidance inside each new issue form.
>
> You can use comments there which will not appear in the final [...]

@Hypfer Great suggestion, i will apply this tommorow. Thanks!

Mimiix, 2020-06-10:
To followup on this, i created a PR:
 https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2913

Kane610, 2020-06-16:
This can be closed now that we have the templates?

Mimiix, 2020-06-16:
Yep, sounds good to me!

closed 2020-06-16 by Mimiix