dresden-elektronik/deconz-rest-plugin #2901: 1.3.7 fehlerhafte Objekte

Xenon-s, 2020-06-08:
Hello, if you delete objects in version 1.3.7 (e.g. "lights" or "groups") the folder is not created again, but only an object "lights" is created. I tested it several times and after a downgrade to 1.3.6 it works. After a new update the error was there again, nothing is in the log.

Mimiix, 2020-06-09:
Goodmorning!
Is this a issue with Phoscon?

Xenon-s, 2020-06-09:
Hello, sorry. No ioBroker in combination with Phsocon. Everything works in Phoscon, but in ioBroker deconz version 1.3.8 is not properly synchronized. If you switch a device in Phsocon, the log also shows as debug that the switching status has changed, but no data point appears in the objects. Downgrade to version 1.3.7 and the objects are already available.

Mimiix, 2020-06-09:
I think you have to ask this question at their GIT repo, as this is not a issue within the deCONZ rest plugin.

closed 2020-06-09 by Xenon-s