dresden-elektronik/deconz-rest-plugin #2902: SmartThings Button Appears in HomeKit but not in Phoscon

fullphat, 2020-06-08:
Hi there!

New to Conbee and decCONZ, but loving it already! 😁 

Using ConBee 2 plugged in to a Raspberry Pi 3.  deCONZ installed and integrated into (separate) Homebridge instance using the `homebridge-hue` plugin.

Today I purchased a Samsung (Samjin) SmartThings button (https://www.samsung.com/uk/smartthings/button-gp-u999sjvleea/) which includes a temperature sensor and, while it all works perfectly and appears in HomeKit, I'm having a few small issues with the Phoscon web app:

1. Only the button appears in the app; the sensor doesn't appear
2. Renaming the button doesn't change HomeKit (homebridge-hue seems to get the device name from the temperature sensor)
3. Battery status is missing from the web app, but appears fine in HomeKit

None of these are major issues, but I thought I'd report them in case it was something I can quickly fix!

Thanks

ebaauw, 2020-06-08:
> Renaming the button doesn't change HomeKit (homebridge-hue seems to get the device name from the temperature sensor)

You can only change the accessory and service name in HomeKit through a HomeKit app.

Mimiix, 2020-06-08:
As per #2897 , Please post phoscon related issues in https://github.com/dresden-elektronik/phoscon-app-beta

I've added the phoscon label. @YKO-de should move it to their repo.

fullphat, 2020-06-08:
Thanks - apologies!

fullphat, 2020-06-08:
> > Renaming the button doesn't change HomeKit (homebridge-hue seems to get the device name from the temperature sensor)
> 
> You can only change the accessory and service name in HomeKit through a HomeKit app.

If I rename a sensor in the Phoscon web app, and then restart Homebridge, it picks up the name change fine.  The issue here seems to be that deCONZ is doing everything right, it's just the web app isn't displaying the sensor and the plugin seems to use that as the name for both.

fullphat, 2020-06-08:
I've recreated this ticket over at Phoscon 👍

closed 2020-06-08 by fullphat