dresden-elektronik/deconz-rest-plugin #2903: Cannot edit some lights in scene

LukeDefeo, 2020-06-08:
I want to alter the CT values of all the lights in a scene. I have a script that fetches all the lights in the scene makes a call for each light. This is the response for fetching the scene info:

GET  http://192.168.0.58:8085/api/KEY/groups/1/scenes/2
```{
    "lights": [
        {
            "bri": 86,
            "colormode": "ct",
            "ct": 50,
            "id": "2",
            "on": true,
            "transitiontime": 0
        },
        {
            "bri": 1,
            "colormode": "ct",
            "ct": 50,
            "id": "3",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 158,
            "colormode": "ct",
            "ct": 232,
            "id": "7",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 158,
            "colormode": "ct",
            "ct": 232,
            "id": "9",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 96,
            "colormode": "ct",
            "ct": 232,
            "id": "17",
            "on": true,
            "transitiontime": 0
        },
        {
            "bri": 187,
            "colormode": "ct",
            "ct": 232,
            "id": "12",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 97,
            "colormode": "ct",
            "ct": 232,
            "id": "8",
            "on": true,
            "transitiontime": 0
        },
        {
            "bri": 158,
            "colormode": "ct",
            "ct": 232,
            "id": "11",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 187,
            "colormode": "ct",
            "ct": 232,
            "id": "10",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 158,
            "colormode": "ct",
            "ct": 232,
            "id": "16",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 187,
            "colormode": "ct",
            "ct": 232,
            "id": "15",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 96,
            "colormode": "ct",
            "ct": 232,
            "id": "14",
            "on": true,
            "transitiontime": 0
        },
        {
            "bri": 158,
            "colormode": "ct",
            "ct": 232,
            "id": "18",
            "on": false,
            "transitiontime": 0
        },
        {
            "bri": 96,
            "colormode": "ct",
            "ct": 232,
            "id": "13",
            "on": true,
            "transitiontime": 0
        }
    ],
    "name": "Evening lights",
    "state": 0
}
```

If i make the following call:

PUT http://192.168.0.58:8085/api/KEY/groups/1/scenes/2/lights/2/state
{
   "ct" : 50
}

It returns 200 OK and I can see the change when i get it back. It also works for light ID 3. If I change Id to 18 for example

PUT http://192.168.0.58:8085/api/KEY/groups/1/scenes/2/lights/18/state
{
   "ct" : 50
}

I get 404 
[
    {
        "error": {
            "address": "/groups/1/scenes/2/lights/18/state",
            "description": "resource, /lights/18, not available",
            "type": 3
        }
    }
]

LukeDefeo, 2020-06-08:
Just to be clear, this appears to be a bug, my version of phoscon is 2.05.76

Mimiix, 2020-06-09:
Hi,

Is there any IKEA lights in here? The CT was disabled in .76 but re-enabled in.77:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases

LukeDefeo, 2020-06-09:
So this was me being stupid, a bank of lights were off at the wall

closed 2020-06-09 by LukeDefeo

ebaauw, 2020-06-09:
> So this was me being stupid, a bank of lights were off at the wall

Still a bug, imho, that `/lights` resources are missing on startup and don't re-appear (are unavailable) when devices are powered down.