dresden-elektronik/deconz-rest-plugin #2904: Problem to upgrade firmware at Conbee

callebridholm, 2020-06-08:
I have a Conbee 1 and old firmware, 260B0500. I want to upgrade to 26350500 and tried many many times with USB connected to Win 10 computer and the GUI. It says it was updated but then after reload it is still old firmware. I then tried GCFFlasher but it fails also but at least I got some logs if someone can find what the problem can be.

`C:\Users\calle\AppData\Local\deCONZ\bin>GCFFlasher -l
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
\\.\COM6         | 0x0403 | 0x6015  | DO00KRAFA  | ConBee

C:\Users\calle\AppData\Local\deCONZ\bin>GCFFlasher -d COM6 -x 3 -f ../firmware/deCONZ_Rpi_0x26350500.bin.GCF
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: ../firmware/deCONZ_Rpi_0x26350500.bin.GCF
ls dev: \\.\COM6 (0x0403/0x6015) sn: DO00KRAFA
ls dev: \\.\COM3 (0x8086/0x1E3D) sn:
Reboot device COM6 (ConBee)
serial port open device
deCONZ firmware version 260B0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
reset FTDI DO00KRAFA (dev: 0)
action: update firmware after 3 ms
serial port open device
STARTIN (390 ms)
STARTING APP
 (406 ms)
STARTING APP
? (486 ms)
STARTING APP
? (2616 ms)
could not sync with bootloader
close serial port

C:\Users\calle\AppData\Local\deCONZ\bin>`

Smanar, 2020-06-08:
Try with adding in the command line " -t 60" and "-R 100"

callebridholm, 2020-06-08:
> Try with adding in the command line " -t 60" and "-R 100", not sure fore the second one.
> 
> But wich one deconz version have you ? Because I don't see the version V3.05 for the flasher. http://deconz.dresden-elektronik.de/win/

Downloaded latest version now and executed the following command with the same error:
C:\Users\calle\Downloads\GCFFlasher_Win_V3_10\GCFFlasher_V3_10>GCFFlasher -d COM6 -x 3 -t 60 -R 100 -f deCONZ_Rpi_0x26350500.bin.GCF

`using firmware file: deCONZ_Rpi_0x26350500.bin.GCF
ls dev: \\.\COM6 (0x0403/0x6015) sn: DO00KRAFA
ls dev: \\.\COM3 (0x8086/0x1E3D) sn:
Reboot device COM6 (ConBee)
serial port open device
deCONZ firmware version 260B0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
reset FTDI DO00KRAFA (dev: 0)
GCF_ResetDeviceDone
action: update firmware after 3 ms
serial port open device
STARTIN (391 ms)
STARTING APP
 (407 ms)
STARTING APP
? (487 ms)
STARTING APP
? (2617 ms)
could not sync with bootloader
close serial port
retry, elapsed 49 seconds
parse commands: GCFFlasher -d COM6 -x 3 -t 60 -R 100 -f deCONZ_Rpi_0x26350500.bin.GCF`

Smanar, 2020-06-08:
On the wiki they are using only 

>GCFFlasher -d COM9 -f ../firmware/deCONZ_Rpi_0x26350500.bin.GCF

For conbee 1

callebridholm, 2020-06-08:
> On the wiki they are using only
> 
> > GCFFlasher -d COM9 -f ../firmware/deCONZ_Rpi_0x26350500.bin.GCF
> 
> For conbee 1, so without my parameter, but without "-x" too

Yes I know I started with that but -x and 3 was showing more output I found in another issue post

Smanar, 2020-06-08:
You have closed deconz before making the update ?

callebridholm, 2020-06-08:
> You have closed deconz before making the update ?
> 
> and idk, if that is normal
> 
> > ls dev: .\COM6 (0x0403/0x6015) sn: DO00KRAFA
> > ls dev: .\COM3 (0x8086/0x1E3D) sn:
> 
> Why there is 2 ports.

Yes it is closed and I also restarted PC without staring deCONZ also. I have no idea why COM3 is presented in this output.

wvuyk, 2020-06-21:
It seems that you might have to need an older version of flasher  check #2546 for this?

callebridholm, 2020-06-29:
> It seems that you might have to need an older version of flasher check #2546 for this?

Thanks a lot, that workaround was solving my problem. My Conbee is now updated to latest FW.

closed 2020-06-29 by callebridholm