dresden-elektronik/deconz-rest-plugin #2906: New device request - ZBT DIMController D0800 Müller Licht

salopette, 2020-06-08:
can you add this Müller Licht remote control please!

[image]
[image]
[image]
[image]
[image]

closed 2020-07-10 by stale[bot]

og9gjz567, 2021-01-11:
Hi, is there any chance to get the remote working? ;)

ebaauw, 2021-04-10:
Duplicate issue.  See #4134 for integration/support progress.