dresden-elektronik/deconz-rest-plugin #2908: Samsung SmartThings Multipurpose sensor (2018) reporting and battery life

NickWaterton, 2020-06-09:
I am running the latest FW (conbee II) and deconz server (2.5.77).

My SmartThings Multipurpose sensors (2018) are working, but I have noticed some odd things.

* The temperature, orientation/vibration and open/close status is reported every 5 minutes.
* Battery status is reported every 45 minutes
* Battery status is reported 3 times (once for each of the sensors).
* battery state is dropping by about 1% per day.

At this rate, the sensor will be out of battery in about 3 months (I was hoping for longer battery life). I am using the sensors as door open/close sensors.

I don't mind the contact state being reported every 5 minutes (its actually useful to me), but is there a way to change the reporting interval of the other sensors? This may not be an issue if it's not the cause of the battery drain.

I have the websocket set to `websocketnotifyall` so I can see all the device updates (whether they change or not).

Why does the battery state get reported for each of the sensors? it's the same battery for all three.

Is there a way to extend the battery life? say by increasing the reporting interval? or disabling some of the sensors (temperature or orientation/vibration)?

I have changed the orientation threshold to 90, to prevent very frequent reports from the orientation sensor (due to door vibrations).

This is my high level log (with nothing happening to the door - ie idle state):
```
2020-06-09 08:18:21,346[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:23:24,854[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:28:28,366[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:33:31,862[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:38:35,373[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:43:38,879[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:48:42,387[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:53:45,466[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 86
2020-06-09 08:53:45,481[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 86
2020-06-09 08:53:45,491[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 86
2020-06-09 08:53:45,976[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 08:58:49,446[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:03:52,915[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:03:53,161[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:03:53,169[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:03:53,175[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:04:29,133[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:04:29,233[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:04:29,239[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:04:29,249[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:08:56,419[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:11:54,875[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:13:59,923[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:19:03,430[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:24:06,933[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:29:10,456[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:34:13,947[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:39:17,470[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:44:20,947[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:49:24,440[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:49:24,456[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:49:24,482[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 09:49:24,699[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:54:27,958[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 09:59:31,461[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:04:34,974[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:09:38,482[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:14:41,967[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:19:45,464[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:24:48,950[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:29:52,443[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:34:55,956[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 10:34:55,968[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 10:34:55,975[     FrontDoor.DoorSensors] INFO|      battery_update: Item frontDoorDoorBattery_ST received update 85
2020-06-09 10:34:56,174[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:39:59,422[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
2020-06-09 10:45:02,947[     FrontDoor.DoorSensors]DEBUG|       sensor_update: Item frontDoorDoorContact_ST (ST) received update CLOSED
[... 194 lines cut ...]
[D 2020-06-09 10:34:55,929] (Main.DeconzClient   ) Received WS data: {
  "config": {
    "battery": 85,
    "offset": 0,
    "on": true,
    "reachable": true
  },
  "e": "changed",
  "id": "9",
  "r": "sensors",
  "t": "event",
  "uniqueid": "28:6d:97:00:01:01:cd:cd-01-0402"
}
[I 2020-06-09 10:34:55,930] (Main.DeconzClient   ) published: /zigbee_status/event/Front Door Sensor/json: {"config": {"battery": 85, "offset": 0, "on": true, "reachable": true}, "e": "changed", "id": "9", "r": "sensors", "t": "event", "uniqueid": "28:6d:97:00:01:01:cd:cd-01-0402"}
[I 2020-06-09 10:34:55,931] (Main.DeconzClient   ) published: /zigbee_status/event/Front Door Sensor/battery: 85
[I 2020-06-09 10:34:55,932] (Main.DeconzClient   ) published: /zigbee_status/event/Front Door Sensor/offset: 0
[... cut ...] nt/Front Door Sensor/on: True
[I 2020-06-09 10:34:55,934] (Main.DeconzClient   ) published: /zigbee_status/event/Front Door Sensor/reachable: True
[D 2020-06-09 10:34:55,935] (Main.DeconzClient   ) waiting for WS data
```
Which seems to be a battery report for each sensor (sensor id 8, 9 and 10). Which makes no sense as it's the same battery for all three sensors in one device.

The main issue is the battery life dropping at a rate of 1% a day, any suggestion on how to reduce the drain?

Thanks.

ebaauw, 2020-06-09:
> Why does the battery state get reported for each of the sensors? it's the same battery for all three.

Because we copied this setup from the Hue API and aren't yet exposing the battery as a separate ZHABattery `/sensors` resource (as we do for e.g. the IKEA FYRTUR).

Please don't confuse the web socket notification for the Zigbee attribute report: it's one report from the sensor to deCONZ, which translates it into three web socket notifications. 

> I don't mind the contact state being reported every 5 minutes (its actually useful to me), but is there a way to change the reporting interval of the other sensors?

Change the reporting configuration in the deCONZ GUI.  See the user manual (under _Help_) on how to do that.

> disabling some of the sensors (temperature or orientation/vibration)?

Not sure if that's possible, but you could disable the corresponding attribute reporting (set the intervals to 65535).

> As you can see the reports are every 5 minutes, but a different 5 minute interval for each sensor.

That's the device firmware, but not uncommon.  I see the same for the Hue motion sensor.  Typically different clusters report independently.

> The main issue is the battery life dropping at a rate of 1% a day, any suggestion on how to reduce the drain?

Reduce the reporting rate, so the device powers its radio less frequently.

Note the nature of these button cell batteries: they report 100% for up to a year, and then fall quite quickly.  Did you install a fresh battery, or is this the battery shipped with the device?

NickWaterton, 2020-06-09:
Thanks for the quick response.

> Please don't confuse the web socket notification for the Zigbee attribute report: it's one report from the sensor to deCONZ, which translates it into three web socket notifications.

Ok, so this is a non-issue then.

> Change the reporting configuration in the deCONZ GUI. See the user manual (under Help) on how to do that.

i will try to find out how to do this, and see if it makes a difference.

> Did you install a fresh battery, or is this the battery shipped with the device?

This is the battery shipped with the device. I understand about battery life, I'm used to seeing them sit at 100%, then months/years later drop off a cliff. This is why I was surprised to see it dropping 1% a day for 2 weeks. I was thinking it would stabilize at 90% or so, but no, it just keeps dropping.

Thanks for the pointers, I'll see what i can do to change the reporting interval, and keep an eye on the battery level. I have some fresh batteries, so if it keeps dropping, I'll swap it out.

NickWaterton, 2020-06-11:
Ok, it's in `binding.cpp`. Unusual piece of code, all the values are hard coded. Might be an idea to use an `.xml` file or a `SQlite3` db file to store values for devices, would make it easier to change stuff.

Anyway, the problem is that the values for `VENDOR_SAMJIN` are not quite right. The code says that the values are from https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy
and while most of them are, some were missed and are instead set to default. one of these is battery reporting interval, which should be:
```
zigbee.configureReporting(zigbee.POWER_CONFIGURATION_CLUSTER, 0x0021, DataType.UINT8, 30, 21600, 0x10)
```
According to the Samsung file, ie 21600 seconds (6 hours), but is instead set to default (60 * 45) = 45 minutes. I don't know why 45 minutes would be a good idea for battery reporting interval, but there it is. the minimum change is also set to 1 (ie 0.5%), which should be 16 (8%).

If you increase the value by more than 1.5 times the original value, then it gets reset back to the default value. I think this does it at line 1199:
```
NodeValue &val = bt.restNode->getZclValue(POWER_CONFIGURATION_CLUSTER_ID, rq.attributeId, bt.binding.srcEndpoint);

if (val.timestampLastReport.isValid() && (val.timestampLastReport.secsTo(now) < val.maxInterval * 1.5))
{
    return false;
}

return sendConfigureReportingRequest(bt, {rq});
```
as `rq` contains the default values, and this is part of the  `POWER_CONFIGURATION_CLUSTER_ID` section, which gets called _a lot_.

The ultimate fix is to add a section like this:
```
else if (sensor && sensor->manufacturer() == QLatin1String("Samjin") && (sensor->modelId() == QLatin1String("motion") ||
                                                                         sensor->modelId().startsWith(QLatin1String("multi")) ||
                                                                         sensor->modelId() == QLatin1String("water"))) // Samsung Sensors
{
    rq.minInterval = 30;           // value used by ST config file
    rq.maxInterval = 21600;        // value used by ST config file
    rq.reportableChange8bit = 16;  // value used by ST config file
}
```
After line 1150 (or somewhere similar).

Thanks for the help in tracking this down.

closed 2020-07-10 by stale[bot]

manup, 2021-01-07:
Good catch!
With new JSON based device description files (DDF) these values can be defined in them per device in future.
Till DDF are available I'll update the `bindings.cpp` to align with the groovy file values.

manup, 2021-01-11:
I'm currently testing the topic with a Samjin button, I don't think attribute reporting is the main problem here.

The reason why the battery is trained so fast is that the Samjin devices on lowest MAC level poll the parent node rather quickly if new messages are available. Roughly every 5–10 seconds.

But they do have a Poll Control cluster to configure this behavior, now I'm checking why it isn't used properly (there was some code for this if I remember correctly).

ebaauw, 2021-01-11:
No kidding!  This is my Samjin `multi` sensor:

[image]

manup, 2021-01-11:
The 0 Check-in Interval is valid though, according to the spec it means disabled.

>A value of 0 indicates that the Poll Control Server is turned off and the poll control server will not check-in with the poll control client.

The default Long Poll Interval of 28 is quite intense though and only useful during setup.