dresden-elektronik/deconz-rest-plugin #2909: New device LLKZMK11LM Aqara Two-way control module Wireless Relay Controller 2 channels

KekieM, 2020-06-09:
please add support LLKZMK11LM

[image]

[image]

[image]

[image]

[image]

[image]

SwoopX, 2020-06-09:
Can you also please provide the first screenshot with the clusters being read? Thanks.

Smanar, 2020-06-09:
It's a new version of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1491 ?

KekieM, 2020-06-09:
first print screen now with read

[image]

Smanar, 2020-06-09:
And you have nothing in the API ?

SwoopX, 2020-06-09:
@Smanar On the first look, it doesn't seem so. However, it does not to be fully functional.

Smanar, 2020-06-09:
And what is missing ? The previous issue is not clear for me, only group problem ?

KekieM, 2020-06-09:
i have 2 functional  lights.  detected as dimable licht, that are not dimable only switching.
a non functioning power sensor and a non functioning consumption

SwoopX, 2020-06-09:
@KekieM which deconz version are you running? If you refer to your screenshots with "dimmable light", ignore it. It what the device exposed itself, so nothing what can be done about it.

The electrical measurement cluster is up to 99.9% useless, if you're referring to this. Measurements, if any, are taken via the Analog Input cluster.

Is the switch itself working?

KekieM, 2020-06-11:
running version 2.05.75 (as a addon on home assitant) 

Switch is Working fine.

Rolf56, 2020-06-14:
Hi, in https://www.zigbee2mqtt.io/devices/LLKZMK11LM.html I saw that the option Interlock allows to inter connect the relays, which will make sure that only one relay is on at a time.

How can I set this option to true? Do I use postman for that and whats the syntax for?

Smanar, 2020-06-14:
I think it's missing ATM.
You can add it in the general.xml file, it's just an xml file, you don't need recompilation

Cluster 0x0010.
Attribute: 0xff06
Data Type: Boolean (0x10)
Value 00 or 01

SwoopX, 2020-06-15:
Ok, taken a closer look now. The device is already supported. However, the finetuning is still missing as far as I can tell.

I'll do the following now (also based on the lack of feedback in the other associated issues):
- Prevent a sensor from being created for electrical measurement cluster. It appears utterly useless and doesn't provide any values constantly. It's a power sensor. That being said, I'd recommend to manually delete the corresponding sensor via API.
- Add a new manufacturer specific attribute for the basic cluster of the device to be able to configure interlock mode. That info has just dripped in lately.

Smanar, 2020-06-15:
@SwoopX are you sure for the cluster ? Analog Input (Basic) ? From zigbee2mqtt it's the Binary Output (Basic)

```
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x40)
        .... ..00 = Frame Type: Data (0x0)
        .... 00.. = Delivery Mode: Unicast (0x0)
        ..0. .... = Security: False
        .1.. .... = Acknowledgement Request: True
        0... .... = Extended Header: False
    Destination Endpoint: 1
    Cluster: Binary Output (Basic) (0x0010)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 115
```

SwoopX, 2020-06-15:
Shit, you're right

KekieM, 2020-06-15:
if it is of any help. I have for a test paired the device with the ZHA intergration. 

that creates the useles sensor but alsow a sensor that seems work. the measurements are off in my opinion but at least power consumption is detected.  

if logging fro that test helps let me know what you need.

SwoopX, 2020-06-15:
Ok, so the attribute has already been integrated and put at the right spot now, thanks Smanar. It is available with the .78 release.

However, I'd still like to see the device's special attribute reporting, also for future reference. In that sense, deconz must be run with activated debugging (`--dbg-info=2`). I don't know in what intervalls that's been send, but aournd 20 mins of data should suffice.

KekieM, 2020-06-18:
I have given the relay to a friend for some testing. Expect a second one te be delivered any time. 

I will create the logging when the second relay is delivered.

SwoopX, 2020-06-18:
The changes mentioned here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2909#issuecomment-643991963 have already been included in the latest beta release.

MadMonkey87, 2020-06-19:
Hi
I am on the .78 beta and just added a brand new aqara relay. The sensor resources look like that:
`    "77": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 3600
        },
        "ep": 1,
        "etag": "e35c081211ee0c7107ece20c4b8e8c74",
        "lastseen": "2020-06-19T16:38:09.927",
        "manufacturername": "LUMI",
        "modelid": "lumi.relay.c2acn01",
        "name": "Consumption 77",
        "state": {
            "consumption": 13,
            "lastupdated": "2020-06-19T16:36:27.219"
        },
        "swversion": "02-27-2019",
        "type": "ZHAConsumption"
    },
    "78": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 3600
        },
        "ep": 1,
        "etag": "52590021b0fe2f2edb26021bb43fd7cd",
        "lastseen": "2020-06-19T16:38:09.927",
        "manufacturername": "LUMI",
        "modelid": "lumi.relay.c2acn01",
        "name": "Power 78",
        "state": {
            "current": 0,
            "lastupdated": "2020-06-19T16:38:00.323",
            "power": -1,
            "voltage": 226
        },
        "swversion": "02-27-2019",
        "type": "ZHAPower"
    },`

Voltage seems to be fine, but
- power is most often reported as -1 but sometimes I've got correct values. Should I completly ignore this property or only the -1 values?
 - current is always 0, I think there is something wrong and the device does not really support this reading
  - consumption should be Wh but to me it looks like this is actually a reading in Watts. Could it be that there is something wrong here?
 - battery should not be there

SwoopX, 2020-06-20:
@MadMonkey87 Have you had that one paired before (prior to version .78)?

I heard that the power should be reported in every now and then correctly but never seen that. I really need a debug log of like 15 mins as requested multiple times for this device.

> power is most often reported as -1 but sometimes I've got correct values. Should I completly ignore this property or only the -1 values?

The device shouldn't created a sensor for that at all any more. Really strange that it happened.

> current is always 0, I think there is something wrong and the device does not really support this reading

Not really surprising if that data is not provided. As said before, I need log data to be sure here.

> consumption should be Wh but to me it looks like this is actually a reading in Watts. Could it be that there is something wrong here?

That is all the same. W is what is measured currently, Wh what has been measured over time. So nothing wrong here.

> battery should not be there

That makes sense.

MadMonkey87, 2020-06-20:
Hey @SwoopX 
I've paired it initially with .77, removed it from the guy and re-paired it with .78. I think something went wrong there because I removed and added the device once again and now I indeed have just one ZHAConsumption sensor:

`{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true,
        "temperature": null
    },
    "ep": 1,
    "etag": "fefb333520b4376ed27922fea9846cd7",
    "lastseen": "2020-06-20T16:04:04.799",
    "manufacturername": "LUMI",
    "modelid": "lumi.relay.c2acn01",
    "name": "Consumption 80",
    "state": {
        "consumption": null,
        "lastupdated": "none"
    },
    "swversion": "02-27-2019",
    "type": "ZHAConsumption",
    "uniqueid": "00:15:8d:00:04:45:45:ff-01-000c"
}`

Just a few notes:
- the sensor does not really seem to update any of it's values
- voltage has been gone - this one worked perfectly before
- according to my measurements the actuall watts should be retrieved from the Analog Input cluster, at least this value gets updated every few seconds and seems to be the watts:
[image]

I'll try to get the logs you need as fast as possible

MadMonkey87, 2020-06-20:
Here are some logs, let me know if you need more. The device you're looking for @SwoopX is "0x00158D00044545FF"

[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4808214/debug.txt)

SwoopX, 2020-06-20:
Taken a look but I'm afraid, the timeframe was too short (5 mins if I'm not mistaken). I'd recommend to try it once more and extend the time to about 20 mins.

What you see in the Analog Input cluster should be the historical consumption. The value you see in the GUI is kWh, but the API reports in Wh, so that value is multiplied by 1000. Values may change more or less in real time, but I'd expect beginning of the 2nd digit after the dot.
I've removed the electrical measurement cluster because people were "complaining" that it reports -1 most almost always. So in that sense it is rather useless. However, with the debug output, I'm hoping to better understand what's going on there internally and where that comes from. I'm hoping to catch a "special" reporting package.
Voltage is indeed gone, as it is part of the electrical measurement cluster.

MadMonkey87, 2020-06-23:
Guess this is what you need? Otherwise I need to turn on logging again;)

```
09:26:54:585 ZCL attribute report 0x00158D00044545FF for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
09:26:54:585 0x00158D00044545FF extract Xiaomi special attribute 0xFF01
09:26:54:585    03 temperature 38 °C
09:26:54:585    05 RSSI dB (?) 16 (0x0010)
09:26:54:585    07 unknown 0 (0x0000000000000000)
09:26:54:585    08 unknown 4644 (0x1224)
09:26:54:585    09 unknown 1025 (0x0401)
09:26:54:585    64 on/off 1
09:26:54:585    65 on/off 1
09:26:54:585    6e unknown 1 (0x01)
09:26:54:585    6f unknown 1 (0x01)
09:26:54:585    94 unsupported tag (data type 0x20)
09:26:54:585    95 consumption 0.123750 (124)
09:26:54:585    96 voltage 2254.854004 (225)
09:26:54:585    97 current 0.051331 (0)
09:26:54:585    98 power 6.529763 (7)
09:26:54:585    9b unknown 0 (0x0000)
```
Btw:
- the analog cluster is not the historical consumption. It also reacts pretty quick to changes (unlike the consumption reported via api, it is unusable to detect if the consumer has started/stopped working for example)
- power & current pretty much match what I am measuring manually (watts and amps)

SwoopX, 2020-06-23:
Sweet, exactly what I was looking for. Do you still have the whole debug output and also the line above containing the whole payload? Would be intestesting to see if there's any wurther attribute reporting from the device, other than that.

> the analog cluster is not the historical consumption. It also reacts pretty quick to changes (unlike the consumption reported via api, it is unusable to detect if the consumer has started/stopped working for example)

So the analog cluster is rather actual power usage than consumption?
I know we have a weakness in storing the actually reported values for API reflection as it deviates from the standard handling. However, what's been send via websocket is correct

SwoopX, 2020-06-23:
@MadMonkey87 Do you use x86 arch?

MadMonkey87, 2020-06-24:
Hey @SwoopX 
No I use ArmV8 on a rpi.

The analog cluster indeed reports the actual power usage and it also reflects the values from the special attributes 1:1.

I've created a more extensive log, hope it helps:)
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4826506/debug.txt)

SwoopX, 2020-06-26:
@MadMonkey87 Thanks! Unfortunately, there wasn't anything more of interest for the device. Anyway I reverted the change which prevented the creation of the power sensor and added some code to hopefully just ignore power values of -1.

Once that code has been merged and released, we could have another log to see if and how the situation has changed.

MadMonkey87, 2020-06-26:
Do you have any idea to get the readings for the special attributes into the api  @SwoopX ?

SwoopX, 2020-06-26:
@MadMonkey87 They're caught by default and responsible for you seeing something different than -1

MadMonkey87, 2020-06-26:
Thx a lot, I am excited to thr next release;)

KekieM, 2020-07-18:
I have my second relay switch available do you need me to test something or create more logging?

SwoopX, 2020-07-19:
Closing this as device is now supported with .79 beta.

ebaauw, 2020-07-25:
There's several error in how these are exposed, see https://github.com/ebaauw/homebridge-hue/issues/741.

SwoopX, 2020-07-25:
At least battery exposure should be resolved with .79

ebaauw, 2020-07-25:
Cool!

Any idea why the `modelid` is copied correctly, but `manufacturername` isn't?

By the looks of it, the device indeed advertises itself wrongly as _Dimmable Light_.

SwoopX, 2020-07-25:
> Any idea why the `modelid ` is copied correctly, but `manufacturername ` isn't?

Nope. The device was already supported but oviously, there was some further tweaking required (battery and power). I assume the corresponding request either got lost during pairing or it is again one of those device specific annoyances.

> By the looks of it, the device indeed advertises itself wrongly as Dimmable Light.

Xiaomi, what should I say. I was really thinking about introducing some hashes over the profile ID + endpoints + clusters, but that's probably still not the silver bullet to have the devices uniquely identifiable and prevent the confusion on the profile ID for `/lights`. The JSON would do, but that's too far away I'm afraid.

Rolf56, 2020-08-05:
Hi **SwoopX**, I installed version .79 on my Win10 computer now and found the new entry for the Interlock in general.xlm in section:	<cluster id="0010" name="Binary Output (Basic)">

The Cluster Info Panel in the deConz UI for section "Binary Output (Basic)" shows the comment only but no values at all. Even if it shows on the device details "0010 Binary Output (Basic)  (13)"

Thank you in Advance

SwoopX, 2020-08-06:
I don't get what you mean, tbh. You need to double click on it and set it to true or false, it's a boolean value. It is not pre-filled when it's not read or cannot be read.

Rolf56, 2020-08-06:
As I wrote, there are no attributes at all shown
[image]

SwoopX, 2020-08-06:
Can you restart deconz and check again?

Rolf56, 2020-08-06:
I restarted deconz and I also executed "Reset to Factory Defaults" on the relay device, still the same on both relays I own.

SwoopX, 2020-08-07:
Hm, what about the other cluster on endpoint 02?

Rolf56, 2020-08-08:
In all other clusters are attributes shown exept the Binary Output (Basic) Cluster of both "Dimmable Light" sections. Even endpoint 02

RIGRIG1, 2020-08-11:
I have the same problem as Rolf56

miniknife88, 2020-08-27:
I have the same problem as Rolf56，Unable to enable Interlock

closed 2020-09-29 by stale[bot]

RIGRIG1, 2020-10-03:
Hello!

Please help me activate interlock option. When I try to set interlock option I have next error: UNSUP_MANUF_GENERAL_COMMAND:
[image]

Mimiix, 2020-10-06:
@RIGRIG1 Please open a seperate issue for this if you can ;)

giovanniferretti, 2020-12-12:
Hi, I'm running on the latest stable version (2.07.01) and I think I'm still unable to retrieve power consumption data via REST API - and hence in Home Assistant. I see this info is reported in cluster 000C "Analog Input (Basic)", attribute id 0x0055 "Present value", but this doesn't get exposed to the REST API. I'm not clear if this is something that can be fixed. Thanks, sorry if I missed something from previous conversations

SwoopX, 2020-12-12:
You seem to have a different device than the one mentioned at the beginning of this threat. Is it indeed this one here?

[image]

giovanniferretti, 2020-12-13:
Same device:

[image]

I've tried to delete all references to this device (both switches and sensors) and re-pair after updating deCONZ to 2.07.01 and indeed I'm now getting some power reading, both using REST API and in Home Assistant.

However, one thing I noted: attribute 0x0055 "Present value" in cluster 000C "Analog input (Basic)" seems to update almost immediately to a variation in power consumption; in Home Assistant it gets several seconds  before the value gets updated. Usually ZigBee devices connected to Home Assistant via deCONZ are very responsive e.g. change in status when switching on/off happens istanteneously - is there any reason behind this lag?

[image]

Is the REST API taking the value for the "power" property directly from field 0x0055? Or are any further computations involved maybe?