dresden-elektronik/deconz-rest-plugin #2921: Lost connectivity to all devices after hassio reboot

mhristache, 2020-06-10:
I am using Conbee2 stick and Deconz plugin on Hassio (Home Assistant) on an RPi3 and all worked nicely until I rebooted the host. After reboot, all my devices are greyed out and cannot be used from Deconz portal or Home Assistant.

I tried to remove and insert back the stick, added an extension cable, used different usb ports but nothing helped.

I have enabled some debug and I get this:
```
9:991 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
01:04:29:992 APS-DATA.indication request id: 170 -> finished
01:04:29:992 APS-DATA.request id: 170 erase from queue
01:04:29:992 ZDP status = 0x00 -> SUCCESS
01:04:29:992 ZDP Mgmt_Lqi_rsp zdpSeq: 37 from 0x00212EFFFF05A38C total: 0, startIndex: 0, listCount: 0
01:04:30:390 0x000B57FFFEB72434 error APSDE-DATA.confirm: 0xD0 on task
01:04:30:390 APS-DATA.confirm id: 125, status: 0xD0 
01:04:30:390 max transmit errors for node 0x000B57FFFEB72434, last seen by neighbors 142 s
01:04:31:066 APS-DATA.request id: 176, addrmode: 0x03, addr: 0x000b57fffeb71863, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
01:04:32:130 0x8557 seems to be a zombie recv errors 6
01:04:32:130 LightNode removed 0x000b57fffeb72434
01:04:32:130 Node zombie state changed 0x000b57fffeb72434
01:04:34:770 Mgmt_Lqi_req zdpSeq: 38 to 0x00212EFFFF05A38C start index 0
01:04:34:770 APS-DATA.request id: 192, addrmode: 0x03, addr: 0x00212effff05a38c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 3 len: 2 tx.options 0x00
01:04:34:784 APS-DATA.confirm id: 192, status: 0x00 SUCCESS
01:04:34:784 APS-DATA.confirm request id: 192 -> confirmed, timeout 28825488
01:04:34:791 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
01:04:34:791 APS-DATA.indication request id: 192 -> finished
01:04:34:791 APS-DATA.request id: 192 erase from queue
01:04:34:791 ZDP status = 0x00 -> SUCCESS
01:04:34:791 ZDP Mgmt_Lqi_rsp zdpSeq: 38 from 0x00212EFFFF05A38C total: 0, startIndex: 0, listCount: 0
01:04:39:570 Mgmt_Lqi_req zdpSeq: 39 to 0x00212EFFFF05A38C start index 0
01:04:39:570 APS-DATA.request id: 213, addrmode: 0x03, addr: 0x00212effff05a38c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 3 len: 2 tx.options 0x00
01:04:39:583 APS-DATA.confirm id: 213, status: 0x00 SUCCESS
01:04:39:583 APS-DATA.confirm request id: 213 -> confirmed, timeout 28828192
01:04:39:590 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
01:04:39:590 APS-DATA.indication request id: 213 -> finished
01:04:39:590 APS-DATA.request id: 213 erase from queue
01:04:39:590 ZDP status = 0x00 -> SUCCESS
01:04:39:590 ZDP Mgmt_Lqi_rsp zdpSeq: 39 from 0x00212EFFFF05A38C total: 0, startIndex: 0, listCount: 0
01:04:41:192 0x000B57FFFEB71863 error APSDE-DATA.confirm: 0xD0 on task
01:04:41:192 APS-DATA.confirm id: 176, status: 0xD0 
01:04:41:192 max transmit errors for node 0x000B57FFFEB71863, last seen by neighbors 152 s
01:04:42:209 0x8A3B seems to be a zombie recv errors 6
01:04:42:209 LightNode removed 0x000b57fffeb71863
01:04:42:210 Node zombie state changed 0x000b57fffeb71863
01:04:43:985 Current channel 15
01:04:43:995 Device TTL 3345 s flags: 0x7
```
Did some googling and found that some people complain about corrupt `zbconf` but it looks ok (I think) on my system: 

```
sqlite> select * from zbconf;
{"apsUseExtPanId":"0x0","curChannel":15,"deviceType":0,"extPanId":"0x212effff05a38c","fwversion":"0x264a0700","macAddress":"0x212effff05a38c","networkKey":"6fb3560c1b40d286f780bef623a31a07","nwkAddress":"0x0","nwkUpdateId":0,"panId":"0x99d6","securityMode":3,"staticNwkAddress":false,"swversion":"2.05.77","tcAddress":"0x212effff05a38c"}
```

I would like to avoid resetting the gateway as I have some outdoor devices which are hard to reach to re-pair. 

Any help appreciated!

Thanks

SwoopX, 2020-06-11:
Have a look at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

mhristache, 2020-06-11:
Thanks for the procedure but it did not help. I ended up resetting the gateway and I am in the process of pairing all the devices again. I tried several host reboots and everything recovered, so hopefully this will not happen again.

closed 2020-06-11 by mhristache

SwoopX, 2020-06-11:
@mhristache Just to gain a better understanding of what you did: you upgraded the firmware to the latest available as described in the arcticle and also went through the leave/join in deconz GUI?

mhristache, 2020-06-15:
@SwoopX I already had the latest firmware so I tried the leave/join procedure several times.