dresden-elektronik/deconz-rest-plugin #2922: Update 100 nodes with the correct firmware

ChrisPrefect, 2020-06-11:
I downloaded the tradfri firmware and opened the STD OATU plugin. But it is impossible to find the correct address to a specific device type and select the right firmware and then flash the device and repeat that for all 98 devices.

How can this be done? This absolutely has do work somehow automatically.

[image]

For reference: Script to download the firmware files: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py

ebaauw, 2020-06-11:
Update to the latest version and the OTAU plugin will jump to the corresponding row when you select a node in the map.

Download the firmware files to `~/otau` and restart deCONZ.  The OTAU plugin will match the file automatically.  Mains powered devices will be updated automatically; for battery-powered devices you need to start the update.

Note that the plugin only updates one device at a time.  Also note that it might take some time, before the device queries the OTAU server for new firmware.

ChrisPrefect, 2020-06-12:
Thanks a lot! That sounds like it could work :-)

How can I update the OTAU plugin?

Is there a risk to flash the wrong firmware or can I safely click the update button on each row in the list?

ebaauw, 2020-06-12:
> How can I update the OTAU plugin?

It’s included in the deCONZ distribution.  Not sure, but I think you need v2.05.75 or later for the new functionality.  Alternatively, you can compile and install the OTAU plugin, just like the REST API plugin.

> This should be the correct folder for the firmware files I guess?

Yes.  Make sure to restart deCONZ after downloading the firmware.  The OTAU plugin will copy the files under a new name on startup, after which it should link them, when the device queries the OTAU server for newer firmware.

> Is there a risk to flash the wrong firmware or can I safely click the update button on each row in the list?

The OTAU plugin won’t start sending the firmware when you select a firmware file with the wrong manufacturer or image.  
While adding support for the Hue firmware file format I managed to send a corrupted firmware file to the Hue light.  The firmware downloaded in full, but the device didn’t switch to it afterwards.  So there seems to be some safeguards on the device as well, at least in Hue lights

> I installed the raspian image about three weeks ago, shouldn't this be the latest version?

Apparently not.  You can check the version under the _Help_ menu.