dresden-elektronik/deconz-rest-plugin #2923: Cannot Flash deconz.

vekexasia, 2020-06-11:
Hello All, I was trying to get this fix for issue #2273, but it looks like the page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help is not helping me much.

- I've downloaded GCFFlasher from [here](https://www.dresden-elektronik.de/download/software.html?file=files/dresden-elektronik/content/downloads/software/gcfflasher-2.11.deb) on a raspberry pi,
- added wiring pi from [here](https://project-downloads.drogon.net/wiringpi-latest.deb)
- donwloaded firmware from [here](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26580700.bin.GCF)

but when I try to issue the following command: `sudo GCFFlasher -d /dev/ttyACM0 -t 60 -f ./deCONZ_ConBeeII_0x26580700.bin.GCF` I get *sudo GCFFlasher -d /dev/ttyACM0 -t 60 -f ./deCONZ_ConBeeII_0x26580700.bin.GCF*.

So I issued `-l` and this is the output:
[image]

Which seems to say it's on `AMA` and not `ACM` so I tried -> no luck.
It also says Raspbee but dmesg -w shows this:
[image]

What am I missing?

For further reference I also tried the windows binary but was unable to launch it due to some dll missing (which I installed by dragging them inside the system32 folder -> still same error)

Mimiix, 2020-06-11:
In your post you put parameter ACM0. In your first screen it mentions AMA0.

Did you try that?

vekexasia, 2020-06-11:
Yes!

vekexasia, 2020-06-11:
Ok managed to fix by using docker. All the other alternatives did not work.

closed 2020-06-11 by vekexasia

Mimiix, 2020-06-11:
@vekexasia Not for me, but for others that might have this issue: Can you elaborate how you did this :)?
Just a simple how-to might help. Or did you follow the guide at Marthocs page?