dresden-elektronik/deconz-rest-plugin #2924: Hive motion sensor not paring

donpearson, 2020-06-11:
I have a hive motion sensor that’s not pairing. 
I’m successfully putting the sensor in to pairing mode (orange flashing light). 
What seems to be happening is it’s starting to pair as the orange flashing light stops as it should when it’s paired but that’s as far as it gets I just get a failed to connect error. 
Any ideas what’s wrong?
Thanks

Mimiix, 2020-06-11:
Hi!

Can you explain on how you run this? Do you use a Raspberry PI? VM? Docker?

What version are you running of deCONZ and what is the firmware version of the conbee/raspbee?

What does it show in the deCONZ gui?

Thanks!

donpearson, 2020-06-11:
> Hi!
> 
> Can you explain on how you run this? Do you use a Raspberry PI? VM? Docker?
> 
> What version are you running of deCONZ and what is the firmware version of the conbee/raspbee?
> 
> What does it show in the deCONZ gui?
> 
> Thanks!

Hi thanks for your fast response. 
I’m running it on the pi4. 
Running version 5.3.5 of deCONZ in home assistant. 
I’m unsure where to fined the firmware version for the conbee 2 in home assistant. 
It just says it failed to connect in the gui.

Mimiix, 2020-06-11:
Hi!

5.3.5 is a rollback to version 2.05.75, after there were bugs in 2.0.77 (5.3.4 in HA).

In 2.05.76 this device was added, hence you can't get it working :(

If you find a way to downgrade the addon to 5.3.4 you should be able to get it working. However, i don't know how as i have no clue how docker works. I hope somebody who reads this can help out.

donpearson, 2020-06-11:
> Hi!
>
> 5.3.5 is a rollback to version 2.05.75, after there were bugs in 2.0.77 (5.3.4 in HA).
>
> In 2.05.76 this device was added, hence you can't get it working :(
>
> If you find a way to downgrade the [...]

Oh that’s a bummer, thanks for the information. 
That explains why I can’t get it working.

donpearson, 2020-06-11:
Anyway I can get my sensors working?

Mimiix, 2020-06-12:
@donpearson The only way is to either:
- Find a way to rollback the HA addon
- Run the device stand alone in ubuntu/windows/docker

donpearson, 2020-06-12:
> @donpearson The only way is to either:
> 
> * Find a way to rollback the HA addon
> * Run the device stand alone in ubuntu/windows/docker

Both options you gave me are not possible then. 
Definitely can’t run it as a stand-alone, has to be as it is. 
I wouldn’t have a clue how to rollback the HA addon, I was hoping I’d get that help here.

Mimiix, 2020-06-12:
@donpearson If you are affraid you cant couple with HA: If the new host device is in the same network,  home assistant can read that. If you need some help with that, i can provide that.

The HA Addon is not managed by us, you're better off asking in their repository in how to make your Addon to 0.77 manually. (which means rolling back to 5.3.4 in their addon version).

donpearson, 2020-06-12:
> @donpearson If you are affraid you cant couple with HA: If the new host device is in the same network, home assistant can read that. If you need some help with that, i can provide that.
>
> The HA [...]

Thanks but I really don’t want to have to run another system as it defeats the point in what I’m doing. 

I didn’t know if there was some code I can put somewhere to get my sensor working?

Mimiix, 2020-06-12:
You could compile your own version then edit that, but then you need the knowledge on the HA Addon. 

I'm not familiar with how the Addon works:/

closed 2020-06-16 by donpearson