dresden-elektronik/deconz-rest-plugin #2925: Trust ZYCT-202 switch support

timmo001, 2020-06-11:
I know there have been a few threads on this, which are all now currently closed due to inactivity or becoming off-topic etc. so I would like to state that I am specifically interested in full support of this switch as a switch with all the groups (or multiple switches for each group).

_As a side note_: those struggling to pair, I removed the battery, reinserted, started the pairing process as a sensor and pressed and held the all lights button for 10 or so seconds until half the less showed. I then replaced the battery again and the device showed as ready in the interface.

At the moment, the only events that get outputted are:

- 1002 - on
- 4002 - off
- 2001 - dimmer up pressed
- 2003 - dimmer up released
- 3001 - dimmer down pressed
- 3003 - dimmer down released

What is missing is separate events (or, which would be better, a native switch) for each of the 5 groups, including the 'all groups' group (if possible/readable)

If these events could be added or, even better added natively to the UI (and in turn, the integration with Home Assistant) it would make this remote much more useful than just a single on, off dimmer switch.

MadMonkey87, 2020-06-12:
I think I've seen another event when switching the groups, let me check when I am at home

MadMonkey87, 2020-06-13:
It appears that you'll need to check the config.group field. 
`
{
  "config": {
    "group": "4399",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "443798a3d27a2cfdd7c3bf62a82b5d53",
  "lastseen": "2020-06-13T17:42:31.512",
  "manufacturername": "Trust",
  "mode": 1,
  "modelid": "ZYCT-202",
  "name": "Switch 75",
  "state": {
    "buttonevent": 4002,
    "lastupdated": "2020-06-13T17:42:31.512"
  },
  "type": "ZHASwitch",
  "uniqueid": "xx:xx:xx:xx:xx:xx:xx:2f-01-1000"
}
`

When you change the group on the switch you do not get an event, but when pressing a button the next time an event is emited to notify you that the group has changed.

For now I've registered the following group ids: 
Group 1: 4399
Group 2: 4400
Group 3: 4401
Group 4: 4402
Group 5: 4403 
All: 4404

Do you get the same values for the groups?

timmo001, 2020-06-14:
OK, so this event only shows the first time a button is pressed on a new group.

I get: 

1. 145
1. 145

MadMonkey87, 2020-06-14:
Interesting that you've got different values. I have 1-2 spares of this device, I'll pair them to see if I get different values.

MadMonkey87, 2020-06-15:
Paired another one and I get the following values for the groups: 42885 - 42890. Seems like every switch has its own:/

Btw: do you get a battery reading?

MadMonkey87, 2020-06-15:
I've seen the group config on some other devices too, does anyone know if there is a more generic meaning of this field? How is it retrieved from the zyct-202?

ebaauw, 2020-06-15:
> When you change the group on the switch you do not get an event, but when pressing a button the next time an event is emited to notify you that the group has changed.

That's how the REST API plugin is supposed to work: it eavesdrop on the command the controller sends to the group, and populates `config.group` to reflect that.

> Paired another one and I get the following values for the groups: 42885 - 42890. Seems like every switch has its own

Again, expected behaviour: Typically, a ZLL switch (which I assume this is, given the presence of the _ZLL Commissioning_ cluster) pick a random group at factory reset.

> OK, so this event only shows the first time a button is pressed on a new group.

What do you mean "new group"?  Does the switch have some selector to act on different groups?  In that case, the sensor isn't supported correctly: it should expose a list of groups in `config.group`, cf. e.g. the Tint remote:
```
{
  "config": {
    "group": "16388,16389,16390",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "0f5fcc8a7506c72542feeee5e1a8020c",
  "lastseen": "2020-06-15T20:16:13.421",
  "manufacturername": "MLI",
  "mode": 1,
  "modelid": "ZBT-Remote-ALL-RGBW",
  "name": "Tint",
  "state": {
    "angle": null,
    "buttonevent": null,
    "lastupdated": "none",
    "xy": [
      0.7,
      0.3
    ]
  },
  "swversion": "20180605-17",
  "type": "ZHASwitch",
  "uniqueid": "00:15:8d:00:03:41:88:92-01-1000"
}
```

> What is missing is separate events (or, which would be better, a native switch) for each of the 5 groups, including the 'all groups' group (if possible/readable)

What does the switch looks like in the GUI?  Does it have different endpoints (one per group) that can be used to map the buttonevents?  Or are the commands to different groups sent from the same endpoint?  In that case, it's going to be very hard to expose different `buttonevent` values for the different groups.  I haven't yet figured out how to do that for the Tint either.

MadMonkey87, 2020-06-16:
> What do you mean "new group"? Does the switch have some selector to act on different groups? In that case, the sensor isn't supported correctly: it should expose a list of groups in `config.group`, cf. e.g. the Tint remote:
> 
> ```
> {
>   "config": {
>     "group": "16388,16389,16390",

I guess that is exactly what is missing! The switch has indeed a selector to act on different groups. The only problem I have to integrate this switch properly into homey is that I do not know what group values are possible per device. If it would be like in your example ("16388,16389,16390") that would be perfect. It looks like there are no different endpoints for different groups:

[image]

ebaauw, 2020-06-16:
> The only problem I have to integrate this switch properly into homey is that I do not know what group values are possible per device.

Basically any valid group address (from 1 to 65534?).  In my experience, a device does pick consecutive groups.

> It looks like there are no different endpoints for different groups

That sucks.  I did some serious hacking to list the groups for the Tint in the right order; the problem is you have no clue what the selector is set at, when you first see a group.  You could first see the first group just as easily as the last one, or any in between.  The REST API plugin needs to see at least the first and the last group, before `config.group` contains all groups, and you can begin to think about exposing different `buttonevent` values for different groups.  The current button map code doesn't cater for this, so you'd need a different, and new, way handling this: lookup the position of the group the command was sent to in `config.group`, and combine that with the commend to derive the `buttonevent` value.

The alternative, exposing a single group in `config.group` and creating rules based on `buttonevent` and `config.group` is worse, since, indeed, you don't know what the group values are going to be (and they change with each device reset).

timmo001, 2020-06-16:
What I'm interested in (apologies if this is what we are talking about here) is the device or devices (for each group based on its number, which can always be renamed) showing up in the deconz UI (which I think means they will also show in the Home Assistant integration - I'll have to look into how this works).

At the moment, there is no way of knowing the device has been added and the only way to get to these events is via the API

> The alternative, exposing a single group in config.group and creating rules based on buttonevent and config.group is worse, since, indeed, you don't know what the group values are going to be (and they change with each device reset).

This sounds like a good idea. As you said, the group only shows when changed. What would be nice is to have this group id in the payload every time

ebaauw, 2020-06-16:
> At the moment, there is no way of knowing the device has been added

Use the API to check whether it's listed under `/sensors`.

> What would be nice is to have this group id in the payload every time

No can do, `buttonevent` is in `state`; `group` is in `config.  You'll get an event for the `state`, preceded by an event for `config`, when the selection has changed.

closed 2020-07-18 by stale[bot]