dresden-elektronik/deconz-rest-plugin #2926: Does deconz run stable over the mini UART?

Markkuuss, 2020-06-11:
Does deconz over the mini UART run stable with the Raspberry Pi or does it have to run over the PL011 UART?

In the past, the Raspberry Pi 3 required Bluetooth to be deactivated (using dtoverlay disable-bt) so that the UART PL011 could be used.

The installation guide (https://phoscon.de/en/raspbee/install#raspbian) now only says that the serial port must be activated in raspi-config. In the end this only sets enable_uart=1, so the mini UART is used.

The PL011 offers some advantages over the mini UART.  Does deconz still work correctly with the mini UART? Or must the 'disable-bt' or 'miniuart-bt' be added in the installation instructions?

SwoopX, 2020-06-11:
Although I cannot give you a definitive answer, maybe that much as an indication: I run deconz on a RPi zero with onboard BT enabled with a Raspbee I, no issues for over a year.

My config is as follows:

```
enable_uart=1
#dtoverlay=pi3-disable-bt
```

Maybe @manup can share more insights.

Markkuuss, 2020-06-12:
> My config is as follows:
> 
> ```
> enable_uart=1
> #dtoverlay=pi3-disable-bt
> ```
Okay, thanks. This configuration would also mean that your RaspBee I runs over the mini UART of the Raspberry Pi Zero W after you have neither deactivated Bluetooth nor switched the Bluetooth function to the mini UART (miniuart-bt).

Thanks for the information.

closed 2020-07-11 by stale[bot]