dresden-elektronik/deconz-rest-plugin #2930: ConBee II Windows driver signature expired

bgoak, 2020-06-12:
The digital signature of file conbee_ii.cat has expired on 27.02.2020.
Workaround: set Windows time before 27.02.2020 and install driver.

manup, 2020-06-17:
For most Win10 installations the default Windows drivers will be used. We have this on the on the todo list now, also the deCONZ installer/bindary should be signed in future.

bgoak, 2020-06-17:
Im using Server 2012 R2 and there was no default driver available.

MattWestb, 2020-07-11:
Do not perhaps to do pleas !!

dazrave, 2021-02-09:
I think i'm having this same issue

Jano1959, 2021-03-25:
Got the same issue, just wanted to change the radio channel.
So unplugged it from my RP4, installed the Desconz app, inserted the Conbee2 into my windows 10, it can only find it as an "Serial port COMX"  :-/

manup, 2025-05-02:
Closing as resolved

closed 2025-05-02 by manup