dresden-elektronik/deconz-rest-plugin #2931: Conbee II firmware update failed - device stuck in bootloop

kje1, 2020-06-12:
Hi,

I'm very new to all of this but today my conbee II had a failed firmware update and the phoscon app is now showing "Firmware: Not connected". I was originally running 0x26480700 and was trying to upgrade to the latest version available.

I'm running deconz and phoscon through the marthoc/deconz docker container (using docker compose) and followed the steps in the guide to update but unfortunately it appears to have failed and the device in now stuck in a boot loop - I've confirmed this both on linux (running ubuntu 18,04 LTS) via GCFFlasher_internal -l which sometimes returns /dev/ttyACM0 and sometimes doesn't. I've also confirmed on a windows machine where the device manager refreshes every couple of seconds and there are audible noises to say a usb device has been disconnected / connected.

I've read through numerous posts on here which detail setting the timeout to be longer, unplugging and plugging back in, restarting the computer, setting max retries to 1000 etc. so I'm not kind of out of ideas of how to fix it. I've read that the stick cannot be bricked because the bootloader is static but I'm unsure as to how I recover it.

Any help would be greatly appreciated!

kje1, 2020-06-12:
Hi again,

I have just used the command systemctl stop ModemManager and then tried to install the firmware again and it is now working. I think this issue can be closed.

Thanks

SwoopX, 2020-06-12:
That is already mentioned in the wiki:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

kje1, 2020-06-12:
I didn't see that line in the wiki originally as it is mentioned under the "update in ubuntu" section. I have installed deconz via docker. The "update in docker" section, preceeds the ubuntu section and has a comment to say "the steps in this guide won't work - refer to this guide (https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware) instead" which has no mention of needing to stop modemManager when using the docker update method.

SwoopX, 2020-06-13:
So, you would suggest an update of marthoc's documentation in this regard?

kje1, 2020-06-13:
Hi. Yes, I think that would be useful for others in the same position. It wasn't immediately clear to me that stopping modemManager applied to my update method using marthoc's

kje1, 2020-06-13:
M

closed 2020-06-13 by kje1