dresden-elektronik/deconz-rest-plugin #2932: Lutron Connected Bulb Top Button (On) does not get correctly handled

schmosby, 2020-06-13:
Hi, 

I noticed when trying to capture events for my Lutron Connected Bulb remote in Node Red, that the top button never receives an event, the other three buttons work though. I'm seeing this in my log entry below in the deconz log saying there is no handler. I found the commit where it was added and I do see it, but I see 0xfe not 0xFE. Is the 0xfe in the code case sensitive?

22:04:20:265 no button handler for: LZL4BWHL01 Remote ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0xFE

https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ca381006cdd85987d4f8cf43b34e0a489429ab17

ebaauw, 2020-06-13:
What system do you run deCONZ on?  Looks like https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2845.

schmosby, 2020-06-13:
Hi, I'm running through the Home assistant add-on (still on 77). Home Assistant is running on Ubuntu 18.04.4 LTS (GNU/Linux 4.15.0-101-generic x86_64)

ebaauw, 2020-06-13:
Yeah, probably the same issue.  Should be fixed in the next release.  In the mean time you could compile and install the REST API plugin yourself.

schmosby, 2020-06-13:
Thanks for reviewing that it might be already fixed, I'll give it a try in the next release

closed 2020-06-13 by schmosby