dresden-elektronik/deconz-rest-plugin #2934: Flash deCONZ firmware for a RaspBee II

nimric, 2020-06-13:
How should I flash deCONZ firmware for a RaspBee II? I had no success with "sudo GCFFlasher_internal -f deCONZ_RaspBeeII_0x26610700.bin.GCF".

Regards

Mimiix, 2020-06-13:
Hi!

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Shows how, depending on the host system.

nimric, 2020-06-13:
HI Mimixx!

I know the page, but for my RaspBee II it was not working with the command for the old RaspBee. Therefore I was wondering if the command is a different now like for the ConBee and the ConBee II.

Mimiix, 2020-06-13:
Hi Nimric,

can you provide me with a error?

nimric, 2020-06-13:
It's working. Seems I did something wrong. Thanks for your time Mimiix!

closed 2020-06-13 by nimric

phwagner82, 2022-09-23:
Hello! 
is there a way to update the firmware in HomeAssistantOS?

Regards
Philipp