dresden-elektronik/deconz-rest-plugin #2936: deConz events not always firing inside of Home Assistant

PingPongLee, 2020-06-13:
I’m struggling with my ConBee 2 dongle in Home Assistant… 

I noticed that the events send my deConz is not always send to Home assistant.

I’ve tried numerous things and now i don’t know what to do anymore. I have tried different extension cords for the ConBee 2 dongle. I’ve tried moving it all to a Synology DS918+ (plenty of power i believe) from a RPI 3B+, I’ve tried changing wifi channels on my local wifi to make sure there is no interference with the zigbee network. I’ve also checked the deconz integration through VNC and saw that the network was all green. There is no more than a few meters between each connected bulb, and when i do try out the remote, the dongle is right beside it. Really doesn’t matter where it is located its the same behavior. I’ve ofcourse also tried other remotes… Same result

See this video which shows me pressing the button, and the events that hits in Home Assistant in the background. Notice the timestamp, and notice that on some presses it doesn't always get there.

https://youtu.be/2SaJStIoF18

What can i do?

PingPongLee, 2020-06-14:
Want to add the following info as it might be relevant

As mentioned i run deCONZ through Home Assistant. In the app the following version info is mentioned:

Version 2.05.75 / 08/03/2020
Firmware 264A0700

Mimiix, 2020-07-06:
@PingPongLee Hi!

Did you already update to the newest version?

PingPongLee, 2020-07-06:
Mimiix - When i created this topic everything was updated to the newest version at the time, but haven't touched any of it since because of its unstability i cant use it. Has a new version been released since which should fix it?

Mimiix, 2020-07-06:
The newest firmware added some fixes. Same goes for the newest version of the software. You can simply update the addon. 

sidenote: You cannot update the firmware trough HA. You need to do a manual firmware update. Instructions are here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

PingPongLee, 2020-07-13:
I took a look at your link for manual update, but I'm not sure where to get the actual update packages (*.bin.GCF) ?

Mimiix, 2020-07-13:
@PingPongLee Check the page, it shows on every method.

PingPongLee, 2020-07-13:
Okay i updated my dongle to the newest firmware version for ConBee II (26580700) in windows, updated everything on my Home Assistant, and plugged in back in there.

It seems to have helped a bit actually. There are still missing presses, but before it would not be registering anything for a few seconds (like in the video) but now it instead just misses a single press now and then. Might just be because i "spam" the buttons when testing.

SwoopX, 2020-07-13:
You are looking at the wrong end of the chain where, simplified, a lot can go wrong in between. As such, we wouldn't care what arrives in HA, but what leaves or is send via deconz REST API. Also, there could be numerous conditions which might prevent a command from being received or send (e.g. interference).

Such tests are better made against the debug output of the REST API and having a look at the corresponding websocket notifications, to be more specific.

closed 2020-08-14 by stale[bot]