dresden-elektronik/deconz-rest-plugin #2938: Lights not reachable and switches not working after firmware update of a RaspBee II

nimric, 2020-06-14:
I updated the firmware of my RaspBee II and after that the lights are not reachable anymore. I had to reset the lights and search for new ones to get the lights working again. Same with the switches.

scholzmichael, 2020-06-15:
Same hbere, after FW upgrade  and deConz upgrade (both manually because there was no upgrade button in phoscon app) no light was raechable anymore. The leave/join network with saving network settings in between did not help. I had to reconnect each light again. pi 3B with raspbee 2.

bernardovalente, 2020-07-02:
Same here, updating on a hassio vm running on unraid didn't work so updated it in windows (in windows it appeared and updated as a conbee but in hassio as a raspbee), plug it back in hassio and all the ikea switches are unavailable but xioami sensors and most ikea buttons seem ok. In vnc viewer displays that i don't have a single connection in the mesh

Sascha-U, 2020-07-22:
I am experiencing same issue. Running Deconz on a Ubuntu VM on ProxMox ( onNUC). All sensors keep runing fine but Hue Lights and Osram Plug keeps disconnecting since updated to new FW. I ordered a new Osram bulb as I thought it had to do with the Hue light but still same issue. I get the devices connected but after some time it losses connection and I cannot reconnect. I have to remove and reset device to get it back into the network. It then shows green connection but after some time it loses connection again.

sorryusernameisalreadytaken, 2020-07-27:
I had some problems with my deconz (docker from marthoc) and my raspbee-I modul some days ago.
Did you try to load a old. config:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1917#issuecomment-660940337

bsheepdiy, 2020-08-26:
I encountered the same issue. I'm not using docker nor HomeAssistant. I'm using ubuntu witch amd64 package.

Step to reproduce : 
1. New install deconz 2.05.80 / Conbee II (firmware 0x26490700)
2. Pair Philips HUE, all is working properly
3. stop deCONZ
4. upgrade manually the firmware to 0x26580700 using GCFFlasher_internal (disconnect USB / reconnect USB)
5. restart deCONZ
**6.Philips HUE bulb is now unavailable.**

using F5 on deCONZ GUI and power cycle the bulb doesn't solve the issue.

Mimiix, 2020-08-26:
@bsheepdiy Any logs? Would like to see  what happens in the background

Also, can you re-pair?

manup, 2020-08-27:
In that case please check if the network is connected. In the deCONZ GUI a green "In Network" should be seen and the new firmware version should be shown in deCONZ Menu > Help > About deCONZ.

The Phoscon App shows the new firmware version under Phoscon App > Settings > Gateway.

**Note:** You never need to reset all devices.

All previous settings are stored in the database and can be recreated on the hidden Zigbee configuration page.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Did the settings have changed here?

bsheepdiy, 2020-09-05:
Sorry, I ve done a full new install ..;
So no log, no more information ...

closed 2020-10-11 by stale[bot]