dresden-elektronik/deconz-rest-plugin #2939: Can't turn on lights via switch after scheduled turn off in case the lights are already turned off

nimric, 2020-06-14:
I have a schedule to turn off the lights. The problem appears if the lights are already off at the time the schedule gets executed. In this case I can't turn them on anymore via the associated switch. I have to turn them on via the app to get the switch working again.

Mimiix, 2020-06-14:
Hi!

Is this a phoscon related request?

nimric, 2020-06-14:
Hi Mimix!

I am not sure. Since Phoscon is just a web based gui for deCONZ to my understanding, I assume that this is a problem related to deCONZ.

Mimiix, 2020-06-14:
Hi nimric

The schedule is created in phoscon? If that's the case, that's the one controlling deconz.

Please let me know, so I can move the issue to phoscon.

nimric, 2020-06-14:
Yes it's created in phoscon. Can you give me the link please after you moved the issue? Thanks!