dresden-elektronik/deconz-rest-plugin #2940: GLEDOPTO GL-C-008S created event flood

bortim, 2020-06-14:
Hello devs and contributors of deconz-rest-plugin,

I am struggeling with a GLEDOPTO GL-C-008S (RGB+CCT).
It creates a flood of events over the websocket.

At the momenet I have two types of event within the WebSocket:
**1. Attribute-Event -> flood / ca. 300 ms** 
```
{"attr":{"lastseen":"2020-06-14T11:07:34.954Z","manufacturername":"GLEDOPTO","modelid":"GL-C-008S","name":"GLEDOPTO-Controller","swversion":"1.0.8","type":"Extended color light","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"}
{"attr":{"lastseen":"2020-06-14T11:07:36.059Z","manufacturername":"GLEDOPTO","modelid":"GL-C-008S","name":"GLEDOPTO-Controller","swversion":"1.0.8","type":"Extended color light","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"}
```
**2. State-Event -> rare // every change on the data**
```
{"e":"changed","id":"1","r":"lights","state":{"alert":null,"bri":2,"colormode":"xy","ct":245,"hue":0,"on":true,"reachable":true,"sat":247,"xy":[0.4941,0.1676]},"t":"event","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"}
```

If I read the WebSocket-API-Documentation, the "attr"- is not specified.

It looks like the reason for the WebSocket-Message is the "lastseen"

**Why is it a problem for me:**
I am using it together with iobroker and I can't update anything within iobroker as the device is updated all the time.
If I use the API together with an curl-script, everything works finde.

Regards
bortim

bortim, 2020-06-14:
Logs of the Attribute-Event:
From my perspective `ZCL got data for node=0x6DD3, cl=0x0006, at=0x0000, status=0x00, type=0x10` results in an update of "lastseen" as the timesamps from the logentry and the websocket-event are the same.
```
11:07:34:440 Node 0x00158D00042D13E8 is known by 0 neighbors, last seen 0 s
11:07:34:844 poll node 00:12:4b:00:1f:6f:ce:e4-0b
11:07:34:844 Poll light node GLEDOPTO-Controller
11:07:34:896 read attributes of 0x00124B001F6FCEE4 cluster: 0x0006: [ 11:07:34:897 0x0000 11:07:34:897 ]
11:07:34:897 add task 2434 type 19 to 0x00124B001F6FCEE4 cluster 0x0006 req.id 95
11:07:34:897 Poll APS request 95 to 0x00124B001F6FCEE4 cluster: 0x0006
11:07:34:920 Node 0x00158D000465A320 is known by 0 neighbors, last seen 0 s
11:07:34:932 Poll APS confirm 95 status: 0x00
11:07:34:933 Erase task req-id: 95, type: 19 zcl seqno: 200 send time 0, profileId: 0x0104, clusterId: 0x0006
11:07:34:954 ZCL got data for node=0x6DD3, cl=0x0006, at=0x0000, status=0x00, type=0x10
11:07:34:954 Node data 0x00124b001f6fcee4 profileId: 0x0104, clusterId: 0x0006
11:07:34:955 0x00124B001F6FCEE4: update ZCL value 0x0B/0x0006/0x0000 after 0 s
11:07:34:957 Websocket 192.168.2.113:2023 send message: {"attr":{"lastseen":"2020-06-14T11:07:34.954Z","manufacturername":"GLEDOPTO","modelid":"GL-C-008S","name":"GLEDOPTO-Controller","swversion":"1.0.8","type":"Extended color light","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"} (ret = 307)

11:07:35:879 Node 0x00158D0004609703 is known by 0 neighbors, last seen 0 s
11:07:35:939 poll node 00:12:4b:00:1f:6f:ce:e4-0b
11:07:35:940 Poll light node GLEDOPTO-Controller
11:07:35:993 read attributes of 0x00124B001F6FCEE4 cluster: 0x0006: [ 11:07:35:993 0x0000 11:07:35:993 ]
11:07:35:994 add task 2440 type 19 to 0x00124B001F6FCEE4 cluster 0x0006 req.id 103
11:07:35:994 Poll APS request 103 to 0x00124B001F6FCEE4 cluster: 0x0006
11:07:36:038 Poll APS confirm 103 status: 0x00
11:07:36:038 Erase task req-id: 103, type: 19 zcl seqno: 201 send time 0, profileId: 0x0104, clusterId: 0x0006
11:07:36:059 ZCL got data for node=0x6DD3, cl=0x0006, at=0x0000, status=0x00, type=0x10
11:07:36:059 Node data 0x00124b001f6fcee4 profileId: 0x0104, clusterId: 0x0006
11:07:36:060 0x00124B001F6FCEE4: update ZCL value 0x0B/0x0006/0x0000 after 0 s
11:07:36:062 Websocket 192.168.2.113:2023 send message: {"attr":{"lastseen":"2020-06-14T11:07:36.059Z","manufacturername":"GLEDOPTO","modelid":"GL-C-008S","name":"GLEDOPTO-Controller","swversion":"1.0.8","type":"Extended color light","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:12:4b:00:1f:6f:ce:e4-0b"} (ret = 307)
```

SwoopX, 2020-06-14:
Have you compiled @ebaauw repository for that? If I'm not mistaken, the timestamp "Z" is only part of his latest PR.

bortim, 2020-06-14:
Hey @SwoopX ,

yes you are right.
I use an own compilation with a pull request from @ebaauw 
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2849
```
git remote add origin https://github.com/dresden-elektronik/deconz-rest-plugin.git
git fetch origin refs/pull/2849/head
```
I used this to get the code of his request.

Thank you for the hint:
> Fully implement /lights resource lastannounced and lastseen

Looks like this is an issue for the iobroker-adapter.

Regards
Markus

bortim, 2020-06-14:
The routine within iobroker-adapter does the following:
1. get Message
2. switch/case for the event type (r-attribute)
3. Case r = 'lights' call "getLightState(lightID)"
4. Get current values of the object by deconz-rest-plugin
5. Overwrite all attributes

So all in all:
The deconz-adapter is not iterating through the attributes within the "state"-event-attribute.
That's why the status in iobroker is updated all the time.

ebaauw, 2020-06-15:
Firstly, it's not the GLEDOPTO per se that creates the flood: each messages received from a device updates `lastseen`.  If you have a small network, maybe with only one light, that gets polled by the gateway almost continuously, causing the "flood".  If you have many lights, you probably still see the same amount of events, but now cycling through the lights.

Please check out the upcoming v2.05.78.  I've made some additional changes since you pulled the code: `lastseen` no longer shows milliseconds, and you should see at most one event per second for the same light.  Still, I agree this is a lot of events.  I'm half considering limiting the amount of events per light to one a minute (and maybe no longer reporting seconds in `lastseen`).  However, that would only limit the amount of messages when you have a few lights.  On larger networks, where each light only gets polled once a minute, it wouldn't make much of a difference.

If we don't issue the event internally, you wouldn't be able to create rules on `lastseen`.  I could suppress `lastseen` from the web socket notifications, but then the API would return different attributes than the event.  Maybe create another setting (cf. `/config/websocketnotifyall`) to switch events for `lastseen` on or off?  Problem with that this should probably be a setting per client session, not a server-wise setting.

You might want to set `websocketnotifyall` to false, so the web socket event only contains the changed attributes.  Maybe that will reduce the load on iobroker, as step 5 would be skipped.  Then again, maybe that will break the adaptor.

closed 2020-07-14 by stale[bot]