dresden-elektronik/deconz-rest-plugin #2941: Losing Hue Lights after connecting and power cicle

easybeat, 2020-06-14:
Hi
I have a problem with Hue Light LLC011.

I connected the light to deconz and it did also make connections to other lights. Then I wanted to move it to another place, unpluged it and pluged it in at the new room. When plugged it in it did start light changing for about 10 seconds. And now it is no longer to control in deconz and phoscon.

This did happen several times with this light.

Any help on this?

Thanks

Beat
[image]
[image]

easybeat, 2020-06-14:
Additional information. It happens if I turn the light off in deconz, wait for 2 minutes and the unplug it.

When I plug it in again it starts light changing for 10 seconds and I think this means it has lost connection.

BUT just writing this. I tried to switch on off and it still works.

Don't understand that behaviour at all!

If somebody has a clue please let me know!

Mimiix, 2020-06-15:
Goodmorning!

Can you provide some information on:

How do you run deCONZ?
What version are you on?
Since when do you see this behaviour?
Is it a conbee/raspbee?

Thanks!

easybeat, 2020-06-15:
Good morning

I run deconz on Raspbee I with version 2.5.64 and I had this behaviour since a while when my system first crashed and I lost all my lights. This was before the workaraound of power cicle all lights to bring them back.

I moved the light last night to another room and this morning it was still working. So no idea what's going on.

Since then I have not touched my Raspbee and Deconz as I was afraid that it will crash again. Yesterday I thought to have a go again and came accross this.

Not sure if it is safe to update to 2.5.77 as my whole house (all lights and switches) runs on it.

Could this be fixed with 2.5.77 or is there stable release planned soon?

Thanks

Mimiix, 2020-06-15:
Hi @easybeat ,

First things first:

Create a backup of deconz, (easiest is to just make a image of the SD as it is, that way you can put it back easiliy). 

After that, i'd recommend to use this version to update:
 http://deconz.dresden-elektronik.de/raspbian/alpha/deconz_2.05.77-raspbian-jessie-alpha_armhf.deb This build is a possible fix for issues on 0.77. 

A lot has changed since 5.64, so we need to be sure it's on the latest version first.

After that, we can start seeking more solutions :)

easybeat, 2020-06-15:
OK thanks. Will try the update maybe on the weekend and let you know...

Mimiix, 2020-06-15:
There's a update comming tonight (beta). Then 0.77 is moved to stable afaik.

closed 2020-07-13 by stale[bot]