dresden-elektronik/deconz-rest-plugin #2943: All sensors disconnected after reboot

hdejongh, 2020-06-14:
Hello:

setup:
dell VEP 1485 with ESXi 6.7
hassio ova deployed
conbee ii passthrough connected to the deconz in hassio
multiple aqara sensors.

after a reboot of the hassio vm all sensors are disconnected.

things i tried:
1) update firmware of the conbee ii to the latest version
2) left the network (through deconz vnc), hit save a couple of times in the network settings and joined the network again
3) re-added some of the sensors (which work my they get renamed back in deconz/phoscon. It looks like Hassio recognizes them as the same sensor so that's hopefull.

Anybody got a fix?

Mimiix, 2020-06-15:
Hi! 

This feels similiar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2918 

Can you try what he did? In short: Stop the Addon. Turn off Auto boot of the addon , Reboot, and manually start HA addon later on.

Thanks!

hdejongh, 2020-06-15:
yes I tried that as well. Didn't work.

hdejongh, 2020-06-15:
btw I got the conbee2 last week. Im pretty sure I rebooted a couple of times without any issue.
Suddenly it started on Saturday morning...

Mimiix, 2020-06-15:
How many devices are in the mesh?
Can you provide a screenshot of the VNC showing the connections?

hdejongh, 2020-06-15:
[image]

here you go!

Mimiix, 2020-06-15:
Ah, there's only end nodes. the zigbee mesh works best if there's some routers (bulbs, sockets, anything thats powered trough electricity instead of a battery).

This would probably be why stuff cuts out.

hdejongh, 2020-06-15:
yes i know. But a sensor 2 meters away from the Hub should work fine right. (the sockets will arrive tomorrow:)...

ill add a philips hue bulb...

Mimiix, 2020-06-15:
It would in theory, but i've seen this happen numerous times and it doesn't work that way unfortunately. Zigbee needs a mesh.

hdejongh, 2020-06-15:
i have added 2 philips hue light bulbs. Ill restart everything and update you here

hdejongh, 2020-06-15:
do you think it will fix it self? Btw i didn't restart ESXi (the USB dongle is configured for passthrough). I only restarted the VM.
They still all connect to the dongle....

Mimiix, 2020-06-15:
Depending on the sensors it will find other routes. Just give it a few days. 

You could always try re-pairing for those that are broken. Maybe they find other paths.

hdejongh, 2020-06-15:
i guess im better of trying by re-adding them. It looks like hassio understands they are the same devices...

Mimiix, 2020-06-16:
Yep! HA most of the time will "Remember" them.

hdejongh, 2020-06-16:
everything keeps working fine, but i didn't restart yet....
ill do that soon enough...

hdejongh, 2020-06-16:
ok it works! thanks a lot!!

closed 2020-07-18 by stale[bot]