dresden-elektronik/deconz-rest-plugin #2945: Add support for Stelpro baseboard thermostat STZB402+ please

biggerfooter, 2020-06-15:
Is is possible to add support for the Stelpro baseboard thermostat STZB402+ ?

Here are the details:
[image]
[image]
[image]
[image]
[image]
[image]

biggerfooter, 2020-06-15:
Read info for basic cluster and thermostat cluster

[image]
[image]

Mimiix, 2020-06-16:
As per request from @SwoopX i'll close this issue.

closed 2020-06-16 by Mimiix