dresden-elektronik/deconz-rest-plugin #2946: Device can't be deleted - SmartThings button

alaghbar, 2020-06-15:
Smarthings button did not pair properly first time. Did reset button many times. It pairs OK but no events detected. Only reports temperature and battery level. I have another identical button pairs OK.
Using deCONZ container and all other devices OK. Button is removed OK in Phoscon app. Then I manually delete the device and all its entities in home assistant. However, I think it is not completely removed in deconz. After deleting it in Phoscon, using VNC viewer, I can still see it under device named "Temperature 28", I don't see anyway to delete this entity.

When I attempt to pair it again (after resetting the button), it pairs OK in Phoscon and does appear back in Home Assistant, but no events when button pushed, double pushed or held. Please help
[image]

Mimiix, 2020-06-15:
Hi!

Can you provide me:
deCONZ version
HA Addon version
Device Product ID (the type/productnumber)

Thanks!

alaghbar, 2020-06-15:
Sure thing;
HA & deCONZ on docker setup so there's no HA Addon.
HA is 0.111.2
deCONZ: 2.05.77
Device: Samsung SmartThings Button / PN GP-U999SJVLEDA
Please let me know if you require any further information.
Thank you!!

closed 2020-07-13 by stale[bot]

loopphoto, 2020-08-10:
I’m also having this problem where the device events don’t appear.

Mimiix, 2020-08-11:
@loopphoto Either create a own issue with related information or continue to debug on Discord :) 

@alaghbar For some reason, i missed this issue. Can you let me know if you figured it out?

SwoopX, 2020-08-11:
The question would be if the REST API delivers button events. If yes, it's not an issue on our side.

alaghbar, 2020-08-11:
@Mimiix Events come OK from SmartThings Button but when name is assigned/edited in Phoscon, it is not updated in deconz portal, it keeps its initial name "Temperature 24". When it is deleted in Phsocon, it stays in deconz. I use its in docker - latest version. Otherwise, events come out OK

Mimiix, 2020-08-11:
@alaghbar use the Rest API to change its name :) Check documentation on that ^^