dresden-elektronik/deconz-rest-plugin #2947: Philips Hue Dimmer Switch

phdelodder, 2020-06-15:
Recently bought a Philips Hue Dimmer Switch and noticed that the the following behaviour occurs: 

Holding the OFF button first sends an 4000 event before sending an 4001 event. Is this normal behaviour?

ebaauw, 2020-06-15:
Yes, see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-wireless-switches.

closed 2020-06-15 by ebaauw

xaviml, 2020-06-16:
Hi @ebaauw,

I got the same behaviour as @phdelodder. However, I have not experience this before. Press and hold were 2 separate actions. What did it change? Is this the same problem as the middle button from the round IKEA controller (E1810) that sends "press" event everytime is held?

Thanks :)

ebaauw, 2020-06-16:
> Press and hold were 2 separate actions.

Yes, x000 on press, x001 after holding for 800ms (in fact, a x001 every 800ms while holding), and x003 on release.  Or x000 on press and x002 on release within 800ms.  This hasn’t changed since I first added support for the Hue dimmer switch.  The only recent change is to report also the hold time.

Note that, afaik, the Hue bridge is the only switch that’s this sophisticated (next to, of course the recently introduced Hue smart button).  It’s the standard that defined the `buttonevent` values, which were later extended for other devices supporting double press, etc.  Hardly any other device issues a x000 on press and no other issues multiple x001 while holding.

> Is this the same problem as the middle button from the round IKEA controller (E1810) that sends "press" event everytime is held?

No it’s not.  The Trådfri remote sends a message on press, and nothing on release, so the message is mapped to 1002.  If you hold it for 10s, it sends a setup message, which is translated to 1001.  This makes no sense to me, as the 1002 has already been send, and you’ll never get a 1003.  Best ignore the 1001, imho it should never have been supported.