dresden-elektronik/deconz-rest-plugin #2948: Rest API connection times out

BinOstCH, 2020-06-15:
Hi.
I got stuck trying to access the REST API at very early stage.
(Raspi Pi 4B with RaspBee 2)
Following the "Getting started" (http://dresden-elektronik.github.io/deconz-rest-doc/getting_started/) I sent a 

> GET https://dresden-light.appspot.com/discover

(using Firefox with RESTClient from the Raspi itself). 
I do get the expected answer:

> [{
>   "id": "00212EFFFF05C142",
>   "internalipaddress": "192.168.2.64",
>   "macaddress": "00212EFFFF05C142",
>   "internalport": 80,
>   "name": "ResusRaspBee",
>   "publicipaddress": "XX.XXX.XXX.162"
> }]

However the next step fails:

> POST https://192.168.2.64:80/api 

with

> {
>     "username": "988112a4e198cc1211",
>     "devicetype": "my application"
> }

With or without unlocking the gateway (using the Phoscon App on the Raspi) the request times out. I do not get any answer. Most of the times a message as follows shows:

> **https://192.168.2.64:80/api uses an invalid security certificate**
> It seems the web site "https://192.168.2.64:80/api" is using an invalid security certificate. You 
> can add an exception in order to visit the site. Please open the web site on a new tab then click 
> on the button Advanced, then click Add Exception... button and click the button Confirm 
> Security Exception in Add Security Exception dialog. After add the web site "https://192.168.2.64:80/api" to the exception list you can use RESTClient continue to test that website again. 

I do not get a chance to do as mentioned above, as there is always a timeout.
I guess the reason must be something rather basic, but I run out of ideas.
Any help or suggestions are appreicated. Thanks. BinO

SwoopX, 2020-06-15:
Unless you haven't running a properly implemented reverse proxy running, that's the source of the problem.

Usually, everything takes place via unencrypted **http**...

BinOstCH, 2020-06-15:
Very basic, indeed. It worked. Thanks a lot!!

Mimiix, 2020-06-15:
And on that bombshell, i close this case :)

closed 2020-06-15 by Mimiix