dresden-elektronik/deconz-rest-plugin #2949: WebApp login fails /  deConz on Remote desktop

BinOstCH, 2020-06-15:
Hi.
Two more probably basic questions, but I did not find any documentation on it.
My configuration: Raspi Pi 4B with RaspBee 2

1. Starting deConz through remote login opens deConz in the display that is attached directly to the Raspberry Pi but not in the remote window. Starting the PhosconApp from this locally displayed deConz opens the PhosconApp in the remote window. Any possibility to open deConz in the remote window? 

2. Starting the Phoscon App out of deConz works fine. (Choosing Gatewayname + Password). However I cannot login at the WebApp using the same Login/Password. Within the 10min after bootup I can reset the password to default at either PhosconApp or WebApp Login. But what is the default? I did find references to a default password but not the pw itself. Loggin in again through Phoscon App allows to change the password. But I never succeed with the WebApp?

Thanks for any hints/help. BinO

Mimiix, 2020-06-15:
Hi!
On 1: I have no answer, as its a VNC. But really depends on your installation method.
on 2: With Webapp you mean the old black/grey layout? username is delight. Password is your own.  If you changed your PW in phoscon, the webapp is the same.

BinOstCH, 2020-06-15:
thanks for the delightful answer. it worked. :-)

Mimiix, 2020-06-15:
I've changed your request to feature request :)

SwoopX, 2020-06-15:
From your vnc, try `sudo systemctl restart deconz-gui`

alex0411, 2020-06-16:
Did you think about switching to Docker based deconz, it comes with VNC ;-)

BinOstCH, 2020-06-17:
@SwoopX : Thanks. But "sudo systemctl restart deconz-gui" unfortunately did not help. deConz GUI still opens on 'local display' only. 
@alex0411 : Thanks for the hint. The remote connection basically works well (Remotedesktop under Win10 - XRDP on Raspi). As I am not (yet) familiar with Docker, I will have a look at it.

closed 2020-07-18 by stale[bot]