dresden-elektronik/deconz-rest-plugin #2950: [REST API] bri_inc as group.action command does not update bri of group

nodefeet, 2020-06-15:
## Describe the bug
Changing the brightness for a group incrementally via bri_inc does not update the brightness of the group

## Steps to reproduce the behavior
API calls via Postman 
1. Set the absolute brightness as start point:
> PUT /groups/1/action body: {"bri": 100}

2. Query brightness of group:
> GET /groups/1 --> response: { ... "bri": 100 ...}

3. Increment the brightness of the group:
>  PUT /groups/1/action body: {"bri_inc": 50}

5. Query brightness of group still gives last absolute value:
>  GET /groups/1 --> response: { ... "bri": 100 ...}

Querying the lights of the group however gives the correct response for the brightness: { ... "bri": 150 ...}

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: (26580700)
 - deCONZ version: (2.05.77)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes

MadMonkey87, 2020-06-15:
Does it work for a single light?

nodefeet, 2020-06-15:
yes it does. Every light of the group has the correct value. Only the group itself does not show the correct value.

nodefeet, 2020-06-15:
Just to make sure my issue is not misleading:
**The group correctly changes its brightness after I send the bri_inc command** 
The only issue is that when I pull the group state bri gets not updated accordingly.

ebaauw, 2020-06-15:
The group state doesn't contain `bri`, only `all_on` and `any_on`.  The group `action` should reflect the last command sent to (a light in) the group, but I don't think deCONZ does that today.  To be honest, I don't know how the Hue bridge reacts to sending `bri_inc` to a group.  I wouldn't be surprise if it doesn't update `action.bri` either.  I wouldn't know what value to update it to.

nodefeet, 2020-06-16:
@ebaauw I see, you're right. 

I overlooked that it's in the action attribute of the group and I kind of understand that you would not update `action.bri` after sending bri_inc.

Thanks!

closed 2020-06-16 by nodefeet