dresden-elektronik/deconz-rest-plugin #2951: DeRestPluginPrivate::clearDb also clears auth table

andreasscherbaum, 2020-06-15:
## Describe the bug

The file database.cpp has a function "DeRestPluginPrivate::clearDb()", and the comment on this function states:
```
Clears all content of tables of db except auth table
```

However in the function code is the following:
```
// clear tables

    const char *sql[] = {
        "DELETE FROM auth",
```

Either the comment is wrong, or outdated.
Or the "auth" table is cleaned by mistake.

Mimiix, 2020-06-15:
As per bug report template, you've removed allot of requested info. 

Please add that info, so we can determine on which version/firmware this is.

andreasscherbaum, 2020-06-15:
I'm sorry, what exactly do you want me to add here?
While working on another issue and looking up the code, I stumbled over this discrepancy.

There is a function which code apparently doesn't match the comment above the function. Can't provide more information than that.

Here is the function: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/database.cpp#L728

Someone else has to see if:
* just leave the discrepancy until someone else discovers it again
* the comment is correct and the table should not be cleaned out
* the code is correct and the code should be fixed

Please advise.

Mimiix, 2020-06-15:
## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II

There's allot of this missing here..

ebaauw, 2020-06-15:
None of that info is relevant for this issue.

andreasscherbaum, 2020-06-15:
@Mimiix Seriously, what has any of the excluded information to do with a discrepancy in the code?
Please review what this report is about. This is not about a specific device, or a supported platform. It's simply a problem in the source code.

Mimiix, 2020-06-15:
Fair enough :)

manup, 2020-06-17:
It looks like the function isn't used at all and seems to be dead code. I think we can safely remove it.

[image]

andreasscherbaum, 2020-07-10:
Thanks for the reminder, I created a PR to remove this function.

PR #3027 (2020-07-10): Remove DeRestPluginPrivate::clearDb()
According to #2951 this function is no longer used
Fix: #2951
commit 8f70162d (2020-07-10): Remove DeRestPluginPrivate::clearDb()

According to #2951 this function is no longer used
Fix: #2951
closed 2020-07-10 by manup