dresden-elektronik/deconz-rest-plugin #2952: Lost all devices twice today

repvik, 2020-06-15:
## Describe the bug
This morning I woke up to a brief power outage. This caused hassio to shit itself, so I reinstalled hassio. When I added Deconz, I was greeted by the password creation screen. Not a good sign.
No devices were paired...

I spent several hours pairing lights, sensors and switches, but after a reboot I was back to scratch. Two sensors remain in Deconz, 60+ other devices are gone. This is the 5th time total I've had all my devices disappear on me. This cannot be expected behaviour?

## Steps to reproduce the behavior
Reboot, apparently.

## Expected behavior
Not losing data on every reboot.

## Screenshots

## Environment

 - Host system: RPI 4 2GB
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (0x26490700)
 - deCONZ version: 2.05.75. 
 - Device: ConBee II
 - Do you use an USB extension cable: yes

I use a _known good_ extension cable, to move the Conbee away from the USB 3.0 SSD that creates noise on 2.4GHz.  

### deCONZ Logs

## Additional context

Mimiix, 2020-06-16:
Hi!

Did it auto-update last night? From 0.77 to 0.75? (Addon 5.3.4 to 5.3.5)?

Kind regards,

morfei1, 2020-06-16:
It's acctually a HA/deCONZ add-on bug. The best way to slove this and have stable zigbee hub in the future is to use a standalone deconz/phoscon hub.

repvik, 2020-06-16:
> It's acctually a HA Supervisor bug, resently. The best way to slove this and have stable zigbee hub in the future is to use a standalone deconz/phoscon installation.

Do you have a link to a bug?

repvik, 2020-06-16:
> Hi!
> 
> Did it auto-update last night? From 0.77 to 0.75? (Addon 5.3.4 to 5.3.5)?
> 

Unknown. The power-outage killed my hass install, so I started from scratch. FW on the conbee was 2.05.75 when I started.

Mimiix, 2020-06-16:
> > Hi!
> > Did it auto-update last night? From 0.77 to 0.75? (Addon 5.3.4 to 5.3.5)?
> 
> Unknown. The power-outage killed my hass install, so I started from scratch. FW on the conbee was 2.05.75 when I started.

The Addon is the 2.05.75 (5.3.5 addon version). Firmware is not that .

However, if you had a power outage, it is not impossible that this also destroyed the addon configuration.

repvik, 2020-06-16:
The power outage may have caused the first config loss, but there was no second power outage - there was just a reboot.

morfei1, 2020-06-16:
> > It's acctually a HA Supervisor bug, resently. The best way to slove this and have stable zigbee hub in the future is to use a standalone deconz/phoscon installation.
> 
> Do you have a link to a bug?

Look for "home assistant supervisor 4.8" in google. You'll find in there forum, lots of peaple having problems using Conbee II + HA with deCONZ add-on that failed resently.

SwoopX, 2020-06-16:
To me it read as if HA doesn't pick up your deconz DB. When being prompted to set a new password, from my experience, that can only happen if the DB has been deleted or got lost.

repvik, 2020-06-16:
> Look for "home assistant supervisor 4.8" in google. You'll find in there forum, lots of peaple having problems using Conbee II + HA with deCONZ add-on that failed resently.

What I find when doing that search is a whole lot of users that have been using the generic device (/dev/ttyAMA0 or ttyUSB0, that apparently has changed with the new hypervisor) instead of the by-id device. I do not have this issue, as I have always used the by-id device as the documentation states.

SwoopX, 2020-06-16:
> Two sensors remain in Deconz, 60+ other devices are gone.

@repvik Literally gone? So they don't appear in deconz GUI any more?

repvik, 2020-06-16:
> > Two sensors remain in Deconz, 60+ other devices are gone.
> 
> @repvik Literally gone? So they don't appear in deconz GUI any more?

Not in deconz gui, not in VNC gui. Gone.

SwoopX, 2020-06-16:
Really strange. Only read about that once, but in that case, the database table schemas were incomplete...

Another user, however, described something similar just recently. Do you still have that condition? If yes, a deconz debug output would come in handy as in his case, it showed that the devices are still there and happily communicating.

repvik, 2020-06-16:
Here's what I see with VNC now:
[image]
Nodes no longer have names, and there's only two of them. Yesterday there were more than 60 devices. 

How do I produce debug output?

repvik, 2020-06-16:
Looking at the regular log output now, it looks a bit strange.

`15:04:14:368 dev /dev/ttyAMA0
15:04:14:368 GW firmware version: 0x26490700
15:04:14:368 GW firmware version is up to date: 0x26490700
15:05:09:342 Current channel 11
15:05:09:350 Device TTL 775 s flags: 0x7
15:06:09:340 Current channel 11
15:06:09:348 Device TTL 714 s flags: 0x7
15:07:09:342 Current channel 11
15:07:09:350 Device TTL 654 s flags: 0x7
15:08:09:337 Current channel 11
15:08:09:345 Device TTL 594 s flags: 0x7
15:08:09:345 Device reset watchdog 7200 s
15:08:14:369 dev /dev/ttyAMA0
15:08:14:369 GW firmware version: 0x26490700
15:08:14:369 GW firmware version is up to date: 0x26490700
15:09:09:343 Current channel 11
15:09:09:351 Device TTL 7140 s flags: 0x7
15:10:09:341 Current channel 11
15:10:09:349 Device TTL 7080 s flags: 0x7
15:10:09:721 Announced to internet https://phoscon.de/discover
15:10:09:721 discovery server date: Tue, 16 Jun 2020 13:10:09 GMT
15:10:09:721 	 local time seems to be ok
15:11:09:343 Current channel 11
15:11:09:351 Device TTL 7020 s flags: 0x7
15:12:09:344 Current channel 11
15:12:09:352 Device TTL 6960 s flags: 0x7
15:12:14:370 dev /dev/ttyAMA0
15:12:14:371 GW firmware version: 0x26490700
15:12:14:371 GW firmware version is up to date: 0x26490700
15:13:09:341 Current channel 11
15:13:09:349 Device TTL 6900 s flags: 0x7
15:14:09:343 Current channel 11
15:14:09:351 Device TTL 6840 s flags: 0x7
15:15:09:341 Current channel 11
15:15:09:349 Device TTL 6780 s flags: 0x7
15:16:09:341 Current channel 11
15:16:09:349 Device TTL 6720 s flags: 0x7
15:16:14:368 dev /dev/ttyAMA0
15:16:14:368 GW firmware version: 0x26490700
15:16:14:368 GW firmware version is up to date: 0x26490700
15:17:09:341 Current channel 11
15:17:09:349 Device TTL 6660 s flags: 0x7`

SwoopX, 2020-06-16:
That's pretty much the right log data but in too less detail and way too less in quantity. It looks like `dbg-info` is set to `1`, please raise it to `2`. Also, a log of around 15 mins would help. It must contain the deconz starting process to come to a more sophisticated conclusion.

Appreciate your follow up.

repvik, 2020-06-16:
Log file:
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4787147/deconz.log)

SwoopX, 2020-06-16:
You devinitively seem to have issues with your database file:

```
15:51:16:593 CTRL [00:21:2e:ff:ff:00:c5:ac] found reference in zll.db
15:51:16:594 CTRL [cc:cc:cc:ff:fe:99:7d:06] not viable in zll.db, skip load cache (101)
15:51:16:594 CTRL [cc:cc:cc:ff:fe:dd:2f:f5] not viable in zll.db, skip load cache (101)
```

That is very unusual and also, no sensor has been loaded or is even available there, respectively. You may want to search for the database (filename zll.db) on the file system and check its contents and file permissions and when it has been created.

repvik, 2020-06-17:
Took a while to locate the zll.db. It's inside the docker container...

closed 2020-07-18 by stale[bot]