dresden-elektronik/deconz-rest-plugin #2954: Features for  Xiaomi Mijia (HonneyWell) smoke detector

alahdal, 2020-06-16:
As discussed [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/152) It would be awesome to have the following possibilities in the smoke detector via the rest API or any other way (I am not expert):

- Test the smoke detector.
- Trigger the siren. This is good, in case one of the detector is triggered I can send the trigger to others.
- Change the sensitivity.

For information, I am connecting all smoke detector with OpenHAB.

SwoopX, 2020-06-16:
Thanks. I need the sensor connected to deconz. First of all, a screenshot from deconz GUI where the node clusters have been expanded would help.

Secondly, I require deconz' debug output to have a look at the "special" communication. For that, run decon with `--dbg-info=2 > xiaomi_debug` to clooect that info and save it in a file. Let it run for about a hour and then share it here, please.

It would also be a great help if would could compile the plugin yourself (if required at all) and exchange some files in your setup (nothing serious). By that, we can properly test it.

alahdal, 2020-06-16:
@SwoopX 
Sorry, this is my first debug. What is the full command line to get the debug?

SwoopX, 2020-06-16:
@alahdal I hope you don't run HA

`/usr/bin/deCONZ --dbg-info=2 > xiaomi_debug`

alahdal, 2020-06-16:
@SwoopX The log is being produced, I can read some familiar data

I cannot share the screenshot, because deCONZ GUI is not able to connect to RaspBee. I tried to fix the problem via 
sudo systemctl stop deconz
udo systemctl disable deconz
sudo systemctl start deconz-gui

still no luck.

I am not familiar with the third request (plugin)

alahdal, 2020-06-16:
@SwoopX here is the log: https://justpaste.it/1qvhb

peer69, 2020-06-16:
I can provide screenshot, hope it helps: 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

peer69, 2020-06-16:
I can provide screenshots, hope it helps (not all data seems to be provided by the sensor): 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-06-16:
Thanks guys. @peer69 your're the man. Now one more screenshot from the node with the clusters extended and all is well.

Checking the log now.

SwoopX, 2020-06-16:
@alahdal I'm afraid that data is completely useless. Deconz hasn't connected to your Conbee/Raspbee (as you already mentioned). What have you done there? :)

peer69, 2020-06-16:
> Thanks guys. @peer69 your're the man. Now one more screenshot from the node with the clusters extended and all is well.
> 
> Checking the log now.

[image]
Thats what you meant?

SwoopX, 2020-06-16:
Indeed, thanks!

alahdal, 2020-06-17:
@SwoopX I was not helpful :). I don't know what is exactly the problem, I have made a clean install of Raspberry and deCONZ, still I cannot run deCONZ from the start menu. After several trials, I found that I should stop the service then start deCONZ via command line only. Yet, I am not able to see the same details as provided by @peer69.

Sorry again.

SwoopX, 2020-06-17:
@alahdal it's ok, just wanted to mention that there was something wrong with your setup. Probably it tried to start a 2nd time. So first a `sudo systemctl stop deconz-gui` and then start it again as previously stated. I assume that already does the trick.

alahdal, 2020-06-17:
@SwoopX I believe now every thing is working fine. At least I can see names, accurate last seen status , etc. Before was nothing.

[image]

alahdal, 2020-06-21:
Here is a log
[xiaomi_debug2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4809776/xiaomi_debug2.txt)

DaveRuijter, 2020-06-21:
Any news on this 😎? I have got a bunch of these smoke detectors and these extra features would absolutely be great. Let me know if you need assistance, like testing/debugging/etc.

alahdal, 2020-06-21:
> Let me know if you need assistance, like testing/debugging/etc. Any news already 😎? I have got a bunch of these smoke detectors and these extra features would absolutely be great.

You may join here

SwoopX, 2020-06-21:
Let's please keep the devices seperated.

@DaveRuijter You may read up from here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1182#issuecomment-647119622 as this also pertains the smoke sensors.

andriej, 2020-07-24:
I'm somehow dropped out of the thread... When setting sensitivity will be available in rest-api?

sveip, 2020-08-02:
@SwoopX, I've ordered a few of these, let me know if any more details are needed to make this fully featured in the rest-api.

leonardpitzu, 2020-11-18:
@SwoopX 
the current deconz version (2.05.88) does not support sensitivity or selftest features.
the sensor i own (JTYJ-GD-01LM/BW) supports:

{
    "config": {
        "battery": 100,
        "on": true,
        "pending": [],
        "reachable": true,
        "temperature": 2900
    },
    "ep": 1,
    "etag": "c94ec440826fcfecaa19376a71df9b33",
    "lastseen": "2020-11-18T16:17Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_smoke",
    "name": "Smoke Sensor",
    "state": {
        "fire": false,
        "lastupdated": "2020-11-18T16:17:55.365",
        "lowbattery": false,
        "tampered": false
    },
    "swversion": "20170314",
    "type": "ZHAFire",
    "uniqueid": "00:15:8d:00:04:1d:8f:5e-01-0500"
}

manup, 2020-11-22:
Repopening the issue since self test and sensitivity aren't working yet.

Currently I'm testing the device here, basic smoke detection works with same output as yours, also same `swversion`.

English manual: https://files.xiaomi-mi.com/files/MiJia_Honeywell/MiJia_Honeywell_Smoke_Detector_EN.pdf

In zigbee2mqtt there is code to trigger the self test:
https://github.com/Koenkk/zigbee2mqtt/issues/420#issuecomment-496949545

**Self test**

Write attribute `0xfff1` uint32  -->  value `50397184`  with Lumi manufacturer code `0x115f`.

I had to change a bit of code to let deCONZ pickup the proper attributes in the Cluster Info panel. After that I could send the special value to the device, and it started beeping. Seems to work.

The attribute is **write only** can't be read back.

[image]

However I'm not sure yet how to integrate that properly into the plugin and REST-API. Timing is crucial, the device is awake every 15 seconds, which could lead to packages being lost since these get thrown away by routers when they can't be delivered within 7 seconds. ... tricky.

**Sensitivity**

This seems to be attribute `0xfff0` with uint64 data type and also Lumi manufacturer code `0x115f`. Note in general.xml we have uint32 as data type for this attribute?! But that is auto corrected by deCONZ on reading the attribute.

Reading it brings:

[image]

I have no clue yet which values are supported.

SwoopX, 2020-11-22:
Hey boss, this is what I had to work with (and also no device)

[image]

manup, 2020-11-22:
The sensitivity low, medium and high values in the snipped are interesting, seems that the `0xfff1` attribute is used for multiple purposes, like a command invocation.

I did some tests:

```
/// 0xfff1 values / commands
0x030100000   self test
0x040100000   low sensitivity
0x040200000   medium sensitivity
0x040300000   high sensitivity

// 0xfff0 values
0x0102000011010001   // initial value
0x0102000011010000   // after writing sensitivity low
0x0102000011010000   // after writing sensitivity medium
0x0102000011010000   // after writing sensitivity high
```

Seems currently we don't have a way to verify and read back the actual sensitivity, or perhaps it is reported in the Xiaomi hourly special attribute.

SwoopX, 2020-11-22:
Smoke density is part of the special report. Have forgotten to add it to the wiki, and now I cannot find it anymore...

tsat-psv, 2020-11-23:
>  Timing is crucial, the device is awake every 15 seconds

Is it possible to re-transmit three times with 5 sec. interval?

nukleuz80, 2020-12-25:
Happy Chrismas at all 🎄

How is the status of sensitivity and controlling the sirene?

MacJawa, 2021-01-23:
Any news guys?
Thanks

slcr, 2021-02-11:
i have 12 of these working right now and changing sensivity and trigger the sirene would be a great addition 💪

Jafalex, 2021-07-16:
Hi team, same story here, would be a really nice addition! Thanks

Prior99, 2021-07-27:
Yes, this would be really great. I am planning on buying some of these. Are there any other smoke detectors with self-test / siren already working that you'd recommend instead?

andriej, 2021-07-27:
I've migrated to z2m and it just works.

RealKanashii, 2021-12-31:
waiting for the merge.