dresden-elektronik/deconz-rest-plugin #2955: Cannot update RaspBee II firmware

si3o, 2020-06-16:
HI I want to update my firmware because I have got a lot of connections issues in Deconz after rebooting the PI. (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help)

At the moment my RaspBee has the firmware version **26520700** but I want to upgrade to **26610700** from 08-Jun-2020 13:27  (http://deconz.dresden-elektronik.de/deconz-firmware/)

**grep_ Rev /proc/cpuinfo** 
_Revision	: a02082
Model		: Raspberry Pi 3 Model B Rev 1.2_

**GCFFlasher_internal -l**
_GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee_ 

**sudo GCFFlasher_internal -d /dev/ttyAMA0  -t 10 -f /home/pi/deCONZ_RaspBeeII_0x26610700_many2one.bin.GCF** 
_GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed_

I work with Deconz GUI and stopped it before trying to upgrade firmware.

I hope you can help me :)

closed 2020-06-16 by si3o

SwoopX, 2020-06-16:
I'd be careful with the many2one firmware as it is experimental... Use that one at your own risk or for testing.

ChrisHae, 2020-06-19:
You have to use /dev/ttyS0 after the -d parameter since this is the path of your RaspBee.

7wells, 2021-10-18:
Hello! I have the same problem, i.e. the flash ends with an error message.

Info about my system/kernel in use (a RaspBerry Pi 2 with RaspBee II module):
```
pi@phoscon:~ $ uname -a
Linux phoscon 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux
```

Thus, I want to see the device path (it's /dev/ttyAMA0 and not /dev/ttyS0 as mentioned in the previous posts):
```
pi@phoscon:~ $ GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
```
(BTW, why are vendor and product IDs null?)

And this, to my understanding, is the correct flash command (with correct device path), but it fails:
```
pi@phoscon:~ $ sudo GCFFlasher_internal -d /dev/ttyAMA0 -t 10 -f deCONZ_RaspBeeII_0x26720700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyAMA0 (RaspBee)
deCONZ firmware version 26720700
Bootloader version 0x00030003, app crc: 0x01BB96DC
flashing 164227 bytes: |================|retry, failed
```

What am I missing? Thanks for your help!