dresden-elektronik/deconz-rest-plugin #2956: Delete gateway or factory reset RaspBee after backup restore Home assistant

smartins4, 2020-06-16:
I've reinstalled a backup of homeassistant on my Raspberry Pi (3B+) and now when I open Phoscon it shows me my old gateway but it is greyed out. 
Added a new gateway but fails to connect to network:
11:36:18:694 failed to reconnect to network try=1
11:36:18:694 wait reconnect 1 seconds
11:36:18:736 dev /dev/ttyAMA0
11:36:18:736 COM: /dev/ttyAMA0 / serialno: 
11:36:18:736 COM: --dev: /dev/ttyAMA0 (RaspBee)
11:36:18:771 dev /dev/ttyAMA0
11:36:18:773 device disconnected reason: 4, index: 0
11:36:19:694 wait reconnect 15 seconds
...

Old (and new) RaspBee deCONZ configuration was pointing to /dev/ttyAMA0

ls -la /dev/tty* returns:
crw-rw-rw- 1 root root 5, 0 Jun 9 18:56 /dev/tty
crw-rw---- 1 root audio 166, 0 Jun 9 18:56 /dev/ttyACM0
crw-rw---- 1 root audio 204, 64 Jun 9 18:56 /dev/ttyAMA0

Issues:
Can’t reach or delete old gateway
New gateway fails to connect to network

Help or ideas are welcome.

SwoopX, 2020-06-16:
Like mentioned before, the permissions are set incorrectly. Must be `root dialout` as described [here](https://phoscon.de/en/support) (section "Troubleshooting for ConBee II with Linux").

Under which user are you running deconz? What does the command `id` output then?

smartins4, 2020-06-16:
~ $ id
uid=0(root) gid=0(root) groups=0(root),0(root),1(bin),2(daemon),3(sys),4(adm),6(disk),10(wheel),11(floppy),20(dialout),26(tape),27(video)

smartins4, 2020-06-16:
log is returning:
...
18:03:52:610 created username: 78501B7F36, devicetype: Phoscon#B2302x1218
18:03:52:693 wait reconnect 7 seconds
18:03:53:443 failed to reconnect to network try=1
18:03:53:693 wait reconnect 6 seconds
18:03:53:736 dev /dev/ttyAMA0
18:03:54:694 wait reconnect 5 seconds
18:03:55:693 wait reconnect 4 seconds
18:03:56:693 wait reconnect 3 seconds
18:03:57:718 don't close database yet, keep open for 900 seconds
18:03:57:718 wait reconnect 2 seconds
18:03:58:693 failed to reconnect to network try=2
18:03:58:693 wait reconnect 1 seconds
18:03:58:739 dev /dev/ttyAMA0
18:03:58:739 COM: /dev/ttyAMA0 / serialno: 
18:03:58:739 COM: --dev: /dev/ttyAMA0 (RaspBee)
18:03:58:754 device disconnected reason: 4, index: 0
18:03:59:693 wait reconnect 15 seconds
18:03:59:737 dev /dev/ttyAMA0
18:03:59:737 COM: /dev/ttyAMA0 / serialno: 
18:03:59:737 COM: --dev: /dev/ttyAMA0 (RaspBee)
18:04:00:693 wait reconnect 14 seconds
18:04:01:694 wait reconnect 13 seconds
18:04:02:693 wait reconnect 12 seconds
18:04:03:693 failed to reconnect to network try=3
18:04:03:693 wait reconnect 11 seconds
18:04:03:736 dev /dev/ttyAMA0
18:04:04:693 wait reconnect 10 seconds
18:04:04:844 reuse recently created auth username: 78501B7F36, devicetype: Phoscon#B2302x1218
18:04:04:844 apikey username: 78501B7F36, devicetype: Phoscon#B2302x1218 already exists
18:04:05:693 wait reconnect 9 seconds
18:04:06:693 wait reconnect 8 seconds
18:04:07:693 wait reconnect 7 seconds
18:04:08:693 failed to reconnect to network try=4
...

SwoopX, 2020-06-16:
@smartins4 Can you verify the answer from here?
https://unix.stackexchange.com/questions/205183/changing-the-dev-tty-permissions-at-startup

smartins4, 2020-06-17:
For the sake of clarification, i'm running home assistant 0.111.3 on hassos 4.9.

my config.txt:
disable_splash=1
kernel=u-boot.bin
dtparam=audio=on
enable_uart=1
dtoverlay=pi3-miniuart-bt

root user added to audio with: adduser root audio

SwoopX, 2020-07-08:
The setting `dtoverlay` looks wrong. From what can be found on Raspbian Github, the setting is called `pi3-miniuart-bt-overlay`. Not sure though what happens if you set a wrong parameter name there.

That was taken from HA guide, right?

SwoopX, 2020-07-09:
What also causes this behaviour is when 2 deCONZ instances run at the same time (typically headless and GUI).

Br3b, 2020-07-30:
hello @smartins4 i just ran into the same problem. Could you please tell me how you solved this? Greetings

edwinoonk, 2020-08-08:
so there is no solution I guess

Mimiix, 2020-08-10:
@edwinoonk  @Br3b If you have a issue similar, feel free to open your own issue. It's free 😄. 

My experience learned me that hardly 2 issues are the same. So please, open your own issue and get it fixed!

closed 2020-08-10 by Mimiix