dresden-elektronik/deconz-rest-plugin #2957: Innr SP120 filled log with "No button map for: SP 120"

Willybits, 2020-06-16:
## Describe the bug

  The device is an Innr SP120 powerplug. The device itself seems to be working just fine. However, the log is filled the 'no button map' messages.

## Environment
I am using a Conbee II connected to an Odroid N2 running HassOS 4.6
Deconz version 5.3.4
ConbeeII Firmware: 26580700

### deCONZ Logs
```
11:49:30:026 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
11:49:30:026 ZCL attribute report 0x00158D000388F5A4 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:35:821 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
11:49:35:821 ZCL attribute report 0x00158D0003886181 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:37:438 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
11:49:37:439 ZCL attribute report 0x00158D0003886065 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:38:449 no button map for: SP 120 ep: 0x01 cl: 0x0702 cmd: 0x0A pl[0]: 000
11:49:38:449 ZCL attribute report 0x00158D0003886065 for cluster: 0x0702, ep: 0x01, frame control: 0x1C, mfcode: 0x1166 
11:49:40:817 no button map for: SP 120 ep: 0x01 cl: 0x0008 cmd: 0x0A pl[0]: 000
11:49:40:817 ZCL attribute report 0x00158D0003886181 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:49:832 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
11:49:49:832 ZCL attribute report 0x00158D0003886181 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:51:008 no button map for: SP 120 ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
11:49:51:008 ZCL attribute report 0x00158D000388F5A4 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
11:49:58:008 no button map for: SP 120 ep: 0x01 cl: 0x0B04 cmd: 0x0A pl[0]: 005
11:49:58:008 ZCL attribute report 0x00158D000388F5A4 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
```

## Additional context

ebaauw, 2020-06-16:
These are standard debug messages and don’t indicate an issue.

Mimiix, 2020-06-26:
As @ebaauw answered and no further response is registered, i will close this issue for now!

closed 2020-06-26 by Mimiix