dresden-elektronik/deconz-rest-plugin #2960: Can't upgrade Conbee II FW

mxbrcom, 2020-06-17:
Hello, I can't upgrade Conbee II firmware on Proxmox VM. All packages are up to date.

It idles at set watchdog ttl status: 0x00 and stops working:

`GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
09:57:34:762 using firmware file: deCONZ_ConBeeII_0x26580700.bin.GCF 
09:57:34:829 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2196024
Reboot device /dev/ttyACM0 (ConBee II)
09:57:35:841 query bootloader v1 ID after 1007 ms
09:57:36:349 close serial after 1515
09:57:37:383 query bootloader v3 ID after 2549 ms
09:57:37:383 TX c081027dffc0
09:57:37:884 close serial after 3050
09:57:37:913 query deCONZ firmware version
09:57:37:915 SLIP RX frame length: 9
deCONZ firmware version 264A0700
09:57:37:919 set watchdog ttl status: 0x00`

I downloaded and startet as follows:

`wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26580700.bin.GCF

sudo GCFFlasher_internal -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26580700.bin.GCF -x 3`

manup, 2020-06-17:
Can you please try following call:

`sudo GCFFlasher_internal -sn DE2196024 /dev/ttyACM0 -f deCONZ_ConBeeII_0x26580700.bin.GCF -x 3 -t 60`

mxbrcom, 2020-06-17:
> Can you please try following call:
> 
> `sudo GCFFlasher_internal -sn DE2196024 -f deCONZ_ConBeeII_0x26580700.bin.GCF -x 3 -t 60`

It stops at the same position:

`sudo GCFFlasher_internal -sn DE2196024 -f deCONZ_ConBeeII_0x26580700.bin.GCF -x 3 -t 60
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
10:59:29:499 using firmware file: deCONZ_ConBeeII_0x26580700.bin.GCF 
10:59:29:563 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2196024
10:59:29:564 search FTDI device: no.0
Reboot device  (ConBee II)
10:59:30:581 query bootloader v1 ID after 1007 ms
10:59:31:082 close serial after 1508
10:59:32:115 query bootloader v3 ID after 2541 ms
10:59:32:115 TX c081027dffc0
10:59:32:617 close serial after 3043
10:59:32:645 query deCONZ firmware version
10:59:32:648 SLIP RX frame length: 9
deCONZ firmware version 264A0700
10:59:32:652 set watchdog ttl status: 0x00`

manup, 2020-06-17:
I'm not sure here, it might be a problem that it is running in the VM (?) and the USB interface doesn't re-attach after the power cycle.

Do you have the chance to try to update it on a native OS?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

mxbrcom, 2020-06-17:
> I'm not sure here, it might be a problem that it is running in the VM (?) and the USB interface doesn't re-attach after the power cycle.
> 
> Do you have the chance to try to update it on a native OS?
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Unfortunately not. I am working on Mac OS and using Deconz with Proxmox on an Intel NUC.

stocker85, 2020-06-17:
I update mine via a ubuntu on vmware and i need to physically reseat the USB when I run the flasher for it to reboot. not sure if it helps for you.

mxbrcom, 2020-06-17:
> I update mine via a ubuntu on vmware and i need to physically reseat the USB when I run the flasher for it to reboot. not sure if it helps for you.

That helped - after reseating the Stick during the flash, it ends up with the following:

`12:02:01:221 using firmware file: deCONZ_ConBeeII_0x26580700.bin.GCF 
12:02:01:307 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2196024
12:02:01:308 search FTDI device: no.0
Reboot device  (ConBee II)
12:02:01:319 query bootloader v1 ID after 5 ms
12:02:01:821 close serial after 506
12:02:02:854 query bootloader v3 ID after 1540 ms
12:02:02:855 TX c081027dffc0
12:02:03:356 close serial after 2041
12:02:03:387 query deCONZ firmware version
12:02:03:389 SLIP RX frame length: 9
deCONZ firmware version 264A0700
12:02:03:394 set watchdog ttl status: 0x00
12:02:15:973 query bootloader v3 ID after 10456 ms
12:02:15:974 TX c081027dffc0
12:02:16:065 close serial after 10548
12:02:19:775 close serial after 14258
12:02:20:804 query bootloader v3 ID after 15287 ms
12:02:20:805 TX c081027dffc0
12:02:21:306 close serial after 15789
failed reset -9223368767481
12:02:21:330 retry, elapsed 41 seconds
12:02:21:436 parse commands: /usr/bin/GCFFlasher_internal.bin -sn DE2196024 /dev/ttyACM0 -f deCONZ_ConBeeII_0x26580700.bin.GCF -x 3 -t 60`

What is now the problem?

manup, 2020-06-17:
The problem with VMs is they are really tricky when USB re-attaches, when I remember correctly sometimes auto insert needs to be enabled separately or perhaps insert the device from Virtual Box/vmware menu while GCFFlasher is still running.

mxbrcom, 2020-06-20:
> The problem with VMs is they are really tricky when USB re-attaches, when I remember correctly sometimes auto insert needs to be enabled separately or perhaps insert the device from Virtual Box/vmware menu while GCFFlasher is still running.
> 
> 

Auto Insert is already enabled. I am unfortunately still not able to upgrade the stick.

Sascha-U, 2020-06-29:
I have the same issue. I am not able to upgrade the Conbee Stick on my VM running on ProxMox. Any idea?

SwoopX, 2020-06-29:
The solution is simple: don't do it in a VM, especially not ProxMox

mxbrcom, 2020-06-29:
> The solution is simple: don't do it in a VM, especially not ProxMox

So what is your solution then?

Mimiix, 2020-06-29:
Use windows or anything else? I'm really sorry @mxbrcom but you can't make your problem deCONZ' problem.

mxbrcom, 2020-06-30:
> Use windows or anything else? I'm really sorry @mxbrcom but you can't make your problem deCONZ' problem.

IT was not my intention to make it decent' problem. I just want a working solution :-)

Mimiix, 2020-06-30:
> > Use windows or anything else? I'm really sorry @mxbrcom but you can't make your problem deCONZ' problem.
> 
> IT was not my intention to make it decent' problem. I just want a working solution :-)

There is a working solution, but VM's are not included due to their hard nature. This is something deCONZ can't do anything about.

Sascha-U, 2020-06-30:
> There is a working solution, but VM's are not included due to their hard nature. This is something deCONZ can't do anything about.

I commented on this as it showed up as a bug so I expected that someone works on a solution. If this is an known error I can live with it. But is there a way to upgrade the stick with Windows without installing a full deCONZ on a Windows PC? Is there a documented workaround for VMs?

SwoopX, 2020-06-30:
The solution is pretty simple and probably got misunderstood: use a native OS for the upgrade. Proxmox can obviously not handle the disconnect occurring during flashing, so that is solely a bug with Proxmox. Also, there might be occasions in daily operations which cause serial communicatio to stall and as a safeguarding measure, the stick resets serial communication via a watch dog (as to my knowledge). This can potentially crash Proxmox again due to the already given reasons.

So don't bang your heads against the wall wasting your time with trying to fix things which cannot be fixed, especially not on our end 😉 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

copperfield74, 2020-07-06:
I've had the same problems. My setup is a Synology NAS with Virtualmanager and an Ubuntu VM. Always had problems updating the firmware. Now I've installed the softeware on my Win10 maschine and the firmware update went smoothly.

closed 2020-08-06 by stale[bot]