dresden-elektronik/deconz-rest-plugin #2962: Support for Tasmota-Zigbee (kind of Device Req.)

kugelkopf123, 2020-06-18:
## Feature request type
Support for Tasmota Zigbee.

## Description
Since recently it is possible to use tasmota not only as a Zigbee coordinator but also as a router or endpoint device. 

This works quite well so far. The only problem is that it is currently not compatible with Phoscon or deconz. 
I have dipped into the code of deconz a bit and it seems to me that the IEEE address is used to identify the manufacturer. 

Tasmota achieves its Zigbee capability by adding an external CC253X module. Most of these modules come from some manufacturer in China. Therefore the IEEE addresses are not the same. 
Is there a way to add Tasmota as a standalone device without the IEEE address being required?

Currently it is possible to define the Manufactor Name and ModelID via Tasmota. 
Since Tasmota offers various possibilities, from 1-4 relays or various sensors, it would be good if the model which is created in Phoscon can do everything, or you can categorize this by using TASMOTA as ManufactorName and different versions of the ModelID, e.g. only for 1 relay, 2 relays, 3... Sensors with relays, only sensors etc. 
That would offer incredibly many possibilities. 

Discussion on Tasmota Issues:
https://github.com/arendst/Tasmota/issues/8399#issuecomment-644925199

## Considered alternatives
No known.

## Additional context
A bit older Screenshot from Tasmota connected with deconz raspbee
[image]

kugelkopf123, 2020-06-22:
Can someone give me more information on this? 

@manup perhaps?

closed 2020-07-25 by stale[bot]

SwoopX, 2020-07-25:
What information is static? Model identifier, manufacturer (both from basic cluster) or manufacturer code from the node info panel?

kugelkopf123, 2020-07-28:
> What information is static? Model identifier, manufacturer (both from basic cluster) or manufacturer code from the node info panel?

The things i could set are: 
#define USE_ZIGBEE_MODELID      "Tasmota"  // reported "ModelId"      (cluster 0000 / attribute 0005)
#define USE_ZIGBEE_MANUFACTURER "Tasmota"      // reported "Manufacturer" (cluster 0000 / attribute 0004)
I ve already tried different sets. Like Plug 01 and Osram as Modelid and Manufacturer but there is no device added in phoscon.

SwoopX, 2020-07-28:
Is the manufacturer code always 0x0000? It would be the missing piece here.

kugelkopf123, 2020-07-28:
> Is the manufacturer code always 0x0000? It would be the missing piece here.

Yes! What would be good to put on there? Maybe I can change something in Tasmota's code to say something different.

SwoopX, 2020-07-28:
You might want to join our Discord, so we can discuss this a bit more interactively ;)

I'd guess providing the ability to join would not be too much of a hassle. I see the more challenging part in determining what capabilities tasmota provides then.