dresden-elektronik/deconz-rest-plugin #2965: Hue Dimmer Switches RWL021 not always working on first key press

evadue, 2020-06-18:
## Describe the bug

Once in a while, when pressing (any) button, nothing happens and the Dimmer's LED lights up solid red. Then, upon the next key press, it works again. It's like the Dimmer is asleep and needs to be waken up.

This happens on all my 5 Hue Dimmer Switches. All running on the latest firmware `6.1.1.28573` (same problem on the previous firmware also).

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: NAS
 - Running method: Marthoc Docker container
 - Firmware version: (26580700)
 - deCONZ version: (2.05.77)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

Mimiix, 2020-06-18:
I've had this too, but recently they stopped doing that. I can confirm this happens, but i don't know how to fix it.

MadMonkey87, 2020-06-19:
I think i had the same issue back in the time the switches were paired with the original bridge

ebaauw, 2020-06-19:
> Once in a while, when pressing (any) button, nothing happens and the Dimmer's LED lights up solid red. Then, upon the next key press, it works again. It's like the Dimmer is asleep and needs to be waken up.

Typically that means one of two things:
1. The dimmer's battery is at critical level.  Note that reading the level of button cell batteries is inherently flawed, so it might report critical one moment and 20% the next;
2. The dimmer doesn't receive an ACK from the coordinator on the attribute report.  Most likely because its parent router has been rebooted or switched off, but could also be due to routing issues on the Zigbee network.  I don't know how the dimmer is supposed to react, but if it's any similar to the Hue motion sensor, it will find a new parent, possibly sending the report as a broadcast in the mean time.

For one the solution is simple: try a fresh battery.  For 2 you'd have to sniff the Zigbee traffic to see what's happening.  I have the impression that the recent versions of deCONZ somehow deteriorated the routing issues (at least for Hue devices): I see my Hue motion sensors blink red more often.

ludipq, 2020-07-18:
Well this is still an issue for me. Happens every once in a while. The batteries are fine (>60%)

Mimiix, 2020-07-20:
@ludipq Can you show me a screenshot of your network?

Maybe some logging on when it occurs?

ebaauw, 2020-07-20:
Again, we need a sniffer log of the Zigbee traffic when this happens.  I don’t think this is a bug in deCONZ.  @ludipq are you sure your Zigbee routers (lights) are powered all the time?

ludipq, 2020-07-21:
> @ludipq Can you show me a screenshot of your network?
> 
> Maybe some logging on when it occurs?

> Again, we need a sniffer log of the Zigbee traffic when this happens. I don’t think this is a bug in deCONZ. @ludipq are you sure your Zigbee routers (lights) are powered all the time?

Yes, they are powered 24/7.

Next time it happens I will take note of the date and time and post the logs :)

Mimiix, 2020-08-12:
@ludipq Did anything happen yet?

closed 2020-09-11 by stale[bot]