dresden-elektronik/deconz-rest-plugin #2966: Paulmann Controller Modell: 371000001 SW: 1400-0001

Rambo79, 2020-06-18:
The Paulmann Controller 371000001 will be detected but controlling the connected lights is not possible.

Mimiix, 2020-06-18:
Hi @Rambo79 !

Would you please stick to the [templates](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new/choose) we've created? This would help others greatly in answering your question. 

I'd like to ask you to either add the details from the template in your next comment on this issue or to close this issue and start a new issue with the [templates](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new/choose).

Thank you!

- Dennis

Rambo79, 2020-06-19:
Made a new Device Request by template

closed 2020-06-19 by Rambo79

Mimiix, 2020-06-19:
@Rambo79 Your my hero :)
Thank you!

Rambo79, 2020-06-19:
> @Rambo79 Your my hero :)
> Thank you!

:-) You made my day !