dresden-elektronik/deconz-rest-plugin #2967: Ikea Tradfri Repeater - Latency

rufinus, 2020-06-18:
## Describe the bug
I recently bought some Ikea Blinds - the work as expected. With the blinds they ship small usb plugable repeaters. (Type E1746)

I thought why not and added them to my zigbee mesh. Adding them in phoscon worked with no problems.

The problems started a few hours later. There was a noticable latency in the whole Zigbee Mesh. Aktivating a light in phoscon could take up to 6 seconds befor the light activated. No matter if the bulb had a connection via the repeater or via a other route. Sometimes the commands even doesnt execute.
the deconz gui showed nothing out of the ordinary.

A few minutes/hours after removing the repeaters the mesh was back at a normal speed.

I dont need the repeaters - so i dont have a problem with not using them - but maybe someone is searching for long latency and this may give some pointers.

## Steps to reproduce the behavior
Add Tradfri Repeater to your Mesh Network and wait until the items start to use them

## Expected behavior
No additional Latency on the network and better coverage from the repeaters.

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 264A0700
 - deCONZ version: 2.05.77
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs
Unfortunatly I can't provide any logs. As the repeater where the last things added I removed them to make sure. But after the removal the problem is gone. And i didn't investigate further.

## Additional context
30 Bulbs, LED Controller or Switches
7 Sensors
2 Remotes

Tradfri Range Extender Version: 2.2.005

WLAN mode disabled in phoscon. 
Dedicated WLAN on channel 6 & 11
Zigbee is on Channel 11 (= WLAN 1 from https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html )

If needed/wanted I can send one of the repeaters to any european address if someone wants to debug it.  (free of charge)

ping @Mimiix (sorry for not sticking to the template - as i dont found anything about this, i tought i write i issue to have others see this so if there are unexpected latencies this is one thing to look out for.)

Smanar, 2020-06-18:
>Unfortunatly I can't provide any logs. As the repeater where the last things added I removed them to make sure. But after the removal the problem is gone. And i didn't investigate further.

On my side I have already see 2 time this kind of bug, one time for an ikea plug and another with a philips bulb, I can't explain the cause, but in both case, the user removed the badly device to make it working again.

Larsn1, 2020-06-19:
I removed the repeaters I have as well and network performance is better. They generally build a connection to nearby devices and they seem to improve the mesh so it would be great to find out what the problem is.

Mimiix, 2020-06-19:
I just had a hunch:
I have a ikea repeater and it works fine afaik. No issues. 

However, i am wondering @Larsn1 : Do you use the USB passtrough on the repeater to power another usb device?

Larsn1, 2020-06-19:
> However, i am wondering @Larsn1 : Do you use the USB passtrough on the repeater to power another usb device?

No, never used the USB port whatsoever

Smanar, 2020-06-19:
And you have this problem with many repeater ?

rufinus, 2020-06-19:
@Smanar 2 Repeaters where active on two different floors in the same network.

Larsn1, 2020-06-20:
Just for the records, I had three

PexJensen, 2020-06-20:
I have 2 off these repeaters in my network for months. No problem regarding latency.

But i'm still on version 2.05.69.

Smanar, 2020-06-20:
But its strange, 1 bad device is possible, but 3 ...
IDK if it will be usefull, but can you try to run deconz in debug mode with the repeater, perhaps we can see timeout error or other thing in them ?

closed 2020-07-18 by stale[bot]