dresden-elektronik/deconz-rest-plugin #2968: Problem pairing sensors with multiple functions

KennethLavrsen, 2020-06-19:
## Describe the bug
After upgrading to 2.05.78 I have huge problems pairing Philips Motion sensors, both indoor and outdoor.

These sensors report Motion, temperature and luminance. When you pair them it is pure random which sensors get loaded in the config. It can be Motion only. It can be Temperature and Luminance but no motion.  I have had to delete and pair the same device more than 10 times before suddenly I am lucky and all 3 features are configured. Once it works it stays.

I have seen it with 3 devices. One new outdoor and two old devices that I can deleted, moved to Philips Hue, firmware upgraded them, deleted them from their, and moved them back. It was hell getting them back again because of this. My outdoor sensor still does not report anything else than motion.

I have many of these same sensors - both indoor and outdoor. Those that I did not touch still work. 

There is something wrong with the process of pairing. It seems like Deconz does not complete the loading of all parameters of a new device any more.

When I pair I am sitting 1 meter from the stick. I have no coverage problems with the Zigbee signals at all as I have taken good care to keep Wifi and Zigbee on different frequencies and my stick is away from the noisy computer.

## Steps to reproduce the behavior
Pair a Philips Motion sensor many times and check each time if all 3 features work. You can see it in Phoscon. If it does not work the temperature is 0 deg C, Lux is 0, and No motion

## Environment
I run deconz on bare metal in Ubuntu. No docker or VM. 

 - Host system: NAS
 - Running method: Ubuntu installed from NAS
 - Firmware version: 26580700
 - deCONZ version: 2.05.78
 - Device: Conbee II
 - Do you use an USB extension cable: Yes - 1 meter

### deCONZ Logs

I did not take any logs

KennethLavrsen, 2020-06-19:
I had an additional sensor - it took 4 pairings to succeed.

I should note that I report this on version 2.05.78.

I cannot say if it was present in the past 3-4 versions. Last time I added a sensor with multiple features was end of 2019. I just want to mention that to not lead into the idea this was introduced in the last version

ChrisPl-GH, 2020-06-19:
A few days ago I registered a Philips Motion Outdoor Sensor SML002 (9290019758) at version 2.05.77 and all three sensors were detected directly.

Type: Philips Motion Outdoor Sensor 9290019758
modelid: SML002
swversion: 6.1.0.25261

Smanar, 2020-06-19:
Why do you delete the device beetween 2 try ?
Have you make tries without deleting the device ?

KennethLavrsen, 2020-06-19:
I tried both with and without delete. Last time I had success I did not delete but I cannot remember if I had success with the others with or without

The GUI interface is a bit of a geek expert interface. How would you exactly finish it on the GUI? I would like to try that next time I need it

Smanar, 2020-06-20:
Yep, right, it s more for advanced user, but when you know the trick, it's easy.

- Set phoscon (and not deconz) in permit join, with "add new ...."
- Go on phoscon, select the node, select the cluster list (the second circle on node title.
- Select the Cluster 0x0000 "Basic", look in the cluster panel, on the attribute panel.
- Press the "Read" button, if the device is battery powered, wake up the device.

And another trick for boring device, during the pairing process, try to make them react, for exemple use the magnet for door sensor, or shake it for vibration sensor, ect ...

MarcelMertens, 2020-07-07:
Got exactly the same issue. Also the tip from Smanar did not help.
After about 10 pairing tries without any luck i opened the battery compartment and saw a second howl next to the "Setup" button which has a little "T" labeled. I hoped that T=transmit means.
So i put the Phoscon app into learn mode, reset the motion sensor by holding the setup button for 10sec (green led on) and then start pressing the "T" button several times after. This steps brought up the LUX value

ebaauw, 2020-07-07:
T stands for Touchlink, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1843.

I think re-pairing the sensor probably did the trick already.

KennethLavrsen, 2020-07-29:
I have not seen any checkins that closes this so pinging the stalebot

KennethLavrsen, 2020-07-30:
Just bought a new Philips sensor and still have the issue The sensor does not register the temperature and lux features.

KennethLavrsen, 2020-07-30:
I have tried to pair 20 times now

I have tried the method @Smanar suggested. It reads information. But does not solve the problem

No matter what I do I end up with a sensor that is detecting Motion but always showing 0 lux and 0 deg C in Phoscon and Unknown in Home Assistant. It does not load all 3 sensors

Battery sensor works. Motion and battery.

I have this type of sensors all over the house that I have paired earlier. And not Deconz has been changed so it no longer correctly loads a sensor with multiple functions. 

This report was about to get closed by the stale bot.

It needs some attention. This is basic functionality that used to work great and is now broken

KennethLavrsen, 2020-07-30:
I tried 3 more times and finally it loaded all 3 sensors. Not reasonable.

I forgot to say that in the middle of the many trials I paired it with a Philips Hue and upgraded the firmware to make sure this was not the root cause but it was same shit after. Until now finally after having been fighting for 4 hours

Mariusthvdb, 2020-08-31:
Please let me add the identical issue is happening on an Ikea motion sensor. No batteries state is setup, while the sensor itself is created.

[image]

Mimiix, 2020-08-31:
@Mariusthvdb This doesn't seem to be deCONZ or its integration. If it is, please open a seperate issue.

KennethLavrsen, 2020-08-31:
@Mimiix

I think it is. What he shows is the view from Home Assistant but the symptom is that he has paired a motion sensor to Phoscon  but the battery sensor does not report any level.

This is similar to what I reported. In my case it is mostly the temperature and lux readings that did not work, until I had repaired the device again and again until it succeeds. You can see it in Phoscon that the sensors show default values. Like 0 degrees Celcius and brightness 0.

And I could not make it work with the deconz-gui work around suggested above. The only thing that worked was to keep trying until success.

Mimiix, 2020-08-31:
@KennethLavrsen Nope, it is not. deCONZ doesn't expose LQI in the rest api. Therefore, it can't be used by HA. That must be ZHA or Z2M. I am not familiar with them so i am not sure.

Don't worry, i have forwarded this issue to devs and awaiting response. That's why i didn't reply on the issue. I can provide you with this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Pairing-Devices

To might troubleshoot further.

Mariusthvdb, 2020-08-31:
sorry. @Mimiix you're correct, I was confused by the thread in the community and followed blindly ;-)

my pardon.

Smanar, 2020-08-31:
Just by curiosity have you tried to increase the timer ?

For xiaomi device there is 2 tricks to have better dectection.
- use the sensor during pairing.
- press the button every second during pairing, to prevent the device sleeping.

SwoopX, 2020-08-31:
Sorry folks for being a bit blunt here but without any log data, this is a pretty theoretical discussion. There's a boatload of open questions and an equal variety of possible causes...

So, to really understand what's happening there, we need to see what's going on.

KennethLavrsen, 2020-08-31:
Have you tried to add a Philips Motion sensor lately? Is this a matter of not being able to reproduce it?

Which logs do you refer to? How do I get them? I run deconz on bare metal installed with apt so I can take logs. But each time I need to reproduce it, I spend an hour to get one of my sensors back again. So I need exact instructions how to get deconz to produce the logs you want.

KennethLavrsen, 2020-08-31:
@smanar. The Philips Motion sensor gets detected very fast. And Phoscon shows that the device is paired. Problem is that a Philips sensor is a motion sensor, a light sensor, and temperature sensor and there is a battery sensor as well. And when I pair it, in most cases at least one of the sensors is not working. Lux is 0 or temperature is 0. It is like the pairing is terminated before all the properties are loaded. It used to work 100% reliable. Each new device I added to Phoscon paired flawlessly. It is about the time I raised this report I replaced some IKEA sensor with more Philips sensors that the problems showed itself. And I have been waiving in front of the motion sensor and pressing buttons. It does not help. Only success I have had is to keep on trying adding, removing, adding, removing, until I am lucky all 4 sensors in the Philips motion sensor actually works. I really wonder if any of you devs have tried it yourself recently.

Mimiix, 2020-09-01:
> I really wonder if any of you devs have tried it yourself recently.

Let's keep things friendly, okay? This doesn't benefit any progression here.

Mariusthvdb, 2020-09-01:
> Just by curiosity have you tried to increase the timer ?
> 
> For xiaomi device there is 2 tricks to have better dectection.
> 
> * use the sensor during pairing.
> * press the button every second during pairing, to prevent the device sleeping.

HI, not sure if this was a response to my post? If so, how should I do that, increase the timer. (not sure if that will make a difference, because the same sick used in the Z2M setup is either setup to pair within 2 minutes, or, via config, allowed to api always. Same difficulty and unreliability).

And, tbh, seems a bit immature to need to do all that, for a simple inclusion of a device.  Coming from dedicated Ikea and Hue hubs, where inclusion is faultless, over and over again, (and not talking 1 or 2 devices here), I had expected a better experience.

I was told to use the Deconz stick so I can use the motion sensors better than using the Hue hub (which is polled each 5 seconds max vs the Deconz which pushes immediately) but so far haven't had much success.

> > I really wonder if any of you devs have tried it yourself recently.
> 
> Let's keep things friendly, okay? This doesn't benefit any progression here.

Yes, we should, always. As a newcomer let me phrase it like this: dont you recognize anything at all in these issues? Seems Kenneth posts are very precise in describing what I see locally, so it would be odd if this wasn't seen at your end? At least some of the trouble if not all?

btw, did switch back and forth between Deconz and Z2M so yes, I disabled and restarted in between switches, and seen both integrations with their respective issues...

Still, hoping this can go forward, because Id really love a very swift set of motion sensors!

Do have another issue: I cant get a set of (yet) unknown) zigbee powerplugs to be paired/included. Do we need to do anything special other than clicking the Unknown button in the add dialog? My switches do start flickering their Led indicating it is polled somehow, but they never are included/found as generic power plugs.

Thanks for having a look!Appreciated.

KennethLavrsen, 2020-09-01:
I was not trying to be unfriendly. It is an important datapoint if my problem can be reproduced by the devs or others or it is something in my installation that is unique and that I can do something about myself. If you did not reproduce it because of priority then that is also a datapoint.

Which logs do you need and how do I make them?

Mimiix, 2020-09-01:
@Mariusthvdb Are you currently on deCONZ or Z2M? That is a big thing. If you are on deCONZ: Please let's proceed here.

Personally, i have forwarded this issue to the devs and are awaiting their response. There is not much for me to contribute other than what i Google and find in other git issues. 

@KennethLavrsen It's okay :) Don't worry, no hard feelings here.  Internet makes things harsh.

If you are able to start deconz with the following parameters to write to a log file? 

--dbg-info=1 --dbg-aps=1 --dbg-error=1

As far as i am concerned: Sensors hardly pair correct the first time. they are a pain, especially when you can't keep them woken. For example: a remote can be kept awake when clicking it's buttons. Your motion sensor don't have those which is a pain in this case.

Let's see what comes from the logs.

SwoopX, 2020-09-01:
Sorry to say that, but the debug output mentioned above is mor might not be detailed enough. Should be 

`--dbg-info=2 --dbg-error=2 > outfile`

so all is written to a file. If you really want to go serious, you may also consider compiling an extended plugin version from my repo that basically should catch all or most zigbee communication to and from the device.

Typically, it shouldn't even be necessary to re-pair a device even if sensors were missed. They usually drip in in another sensor search or if you either read the simple descriptors or basic cluster attributes while sensor search is active and the device is awake. To waken it, you need a short push on the physical button or eventually require a needle (iirc, for the Philips indoor).

KennethLavrsen, 2020-09-01:
I will try the debug command way in near future. I did not want to break my installation so I went out and bought an extra Philips motion sensor this afternoon so I can safely add and remove it without causing domestic tension. Happy wife, happy life you know. Give me a day or two

KennethLavrsen, 2020-09-02:
I have been trying this morning with logging and pairing new device

I have seen failure but different failure mode. I have to go to the office now and will continue trying again later today

The failure mode I have reproduced twice it that the sensor is paired but immediately unreachable. Distance between sensor and Conbee II is 1 meter and my 99 other devices work fine. Not an RF problem.

First time I repaired the device (not deleted but repaired) and then it started working except Lux. Lux repaired itself within the duration of a half hour meeting had had.

So I deleted it from Phoscon and repaired. An again the pairing seems to go well with the green OK screen and flashing lights. And again the device is unreachable. I had to stop there to go to work.

The current state with logs are on my wiki.

http://www.lavrsen.dk/foswiki/bin/view/Kenneth/DeconzMultisensorLogs

I will add my steps and more logs to that page. The last log also includes the earlier so you only need to download the latest. I did not have time to truncate the files and they are still manageble sizes.

KennethLavrsen, 2020-09-02:
Supplementary. Since I reported this I have been keeping my Deconz up to date and I am currently on 2.05.80. Firmware on Conbee II is 26580700

KennethLavrsen, 2020-09-02:
Returned from work. Since 11:13 (now 18:41) the Motion sensor is now reachable and fully works.

I understand that a window sensor or temperature sensor may take time to wake up. But a Motion sensor wakes up when you move in front of it and it does that again and again with a short timeout. I cannot believe it is the Philips sensor not sending data. During the pairing something does not complete and either the Conbee, Deconz or the device is hanging in limbo. I have 11 Philips Hue Motion sensors and once they are added and working they are the most reliable Zigbee devices I have. It is only an init problem when you pair them. 

I will try and remove it and pair it again. I will start a new log file as it is beefy now

The log of the entire day has been added to http://www.lavrsen.dk/foswiki/bin/view/Kenneth/DeconzMultisensorLogs. I added a zipped version of the log also

ebaauw, 2020-09-02:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3226#issuecomment-685876007.

Like most Zigbee devices (all except Xiaomi?), the Hue motion sensor doesn’t send any data, until it’s told to do so.  This is done by binding the server (blue) clusters to the gateway and setting up the reporting configuration for the attributes.

> It is only an init problem when you pair them.

Yes, the pairing logic is cursed.  Don’t reset the sensor when re-pairing it, or the whole process restarts from the beginning.  Just search for devices and read the _Basic_ cluster attributes to “pickup” the process where it stalled.  You might need to read the attributes of the clusters corresponding to the mute sensors as well.  Alternatively, setup the bindings and reporting configuration manually in the GUI.  See the user manual (under the help menu).

KennethLavrsen, 2020-09-26:
Go away bot

Mimiix, 2020-09-28:
@ebaauw What do we need to get this fixed?

WhistleMaster, 2020-10-04:
I have the same problem: I have a few difficulties with those as well. The pairing process is quite inconsistent to say the least.

closed 2020-11-09 by stale[bot]