dresden-elektronik/deconz-rest-plugin #2969: Paulmann WW/CW Controller Modell: 371000001

Rambo79, 2020-06-19:
## Device

- Product name: Paulmann Controller 371000001
- Manufacturer: Paulmann Licht
- Model identifier: 371000001
- Device type:
  - [X] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Screenshots
[image]
[image]
[image]

SwoopX, 2020-06-20:
It should work right out of the box. Doesn't it?

Rambo79, 2020-06-20:
I can found the controller (have 2 of them) but can‘t control the connected lights. The wirering is correct (tested with another controller)

SwoopX, 2020-07-07:
So, when you select the `on/off` cluster and then toggle the light, it doesn't work?

Rambo79, 2020-07-08:
@SwoopX No, nothing happens :-(

Smanar, 2020-07-09:
And if you try the level control ?

Rambo79, 2020-07-09:
@SwoopX The Same. My Conbee is integrated in ioBroker and i tried it with every data point. No Function at all

Smanar, 2020-07-09:
The answer was for me ?
So even the level control cluster using transition time not working ?

SwoopX, 2020-07-09:
Please try it directly via deconz' REST API, so not with ioBroker

Rambo79, 2020-07-10:
@Smanar 
> The answer was for me ?
> So even the level control cluster using transition time not working ?
> 
> You have error message or no result ?

Sorry, hit the wrong name on the mobile phone display.

Again no result by controlling the level by Rest-Api :-(

It says "success" but no light goes on

SwoopX, 2020-07-10:
This all doesn't fit together. Can you reset the light and re-pair it?

Rambo79, 2020-07-25:
@SwoopX 
Sorry for m late response. I had to do something different the last few days. Your right. After a reset (which is a little bit complicated) and new pairing, it works. I will close the request. Thanks!!! for any help.

closed 2020-07-25 by Rambo79