dresden-elektronik/deconz-rest-plugin #2970: trouble on my - Blitzwolf temp and humidity sensor BW-IS4

rix30, 2020-06-19:
Hi, I'm newbye in deconz so I try to write clearly, thank you in advance for your help.
I've seen in other resolved issue that Blitzwolf IS4 is  supported from deconz if FW and app are updated, I've  a Conbee II updated to FW 26580700,  deconz updated too to version 2.05.77.
This is all I can see in vnc:
[image]

I can't see the device in webapp nor in home assistant. What am I supposed to do to use it?

THX Riccardo

SwoopX, 2020-06-21:
Try re-pairing it. Might not always work on the first take. Just to prepare you: it will not show up in Phoscon.

rix30, 2020-06-24:
Thank you SwoopX, I've tried to re-pair it a million of times, it always  appears as in the image, and nothing in home assistant, even if I force resync. I'm thinking to change way to create my thermos regulation.

SwoopX, 2020-06-24:
Can you please read the attributes of the basic cluster? Make sure the device is awake, so e.g. take it in hand and/or warm it.

rix30, 2020-06-26:
just yesterday may raspberry stopped working due to a SD fault, now I'm reinstalling all because my backup is too old and deconz is missing.  I'll answer here as soon as the system will be online.
Thank you

rix30, 2020-07-02:
Here I am, with a new SD and the old configuration with the same problem. @SwoopX you've asked me to read the basic cluster, I don't know how to do, I've seen other reading it and posting in the forum, but I don't know where to see. 
Thank you

SwoopX, 2020-07-07:
[image]

rix30, 2020-07-08:
I did it, I've lost cluster window, now I've found it under under. It gave this:
[image]

SwoopX, 2020-07-08:
Thanks. Now do the same when sensor search in Phoscon is active, please. The name of the node should change then.

rix30, 2020-07-08:
YES!!! It works! Thank you a lot! Now I can see it in HA too. You rock!
I close this issue.
Riccardo

closed 2020-07-08 by rix30

SwoopX, 2020-07-08:
I know 😉 

Enjoy!