dresden-elektronik/deconz-rest-plugin #2971: Paulmann RGBW light

cma-bw, 2020-06-19:
## Device

- Product name: Smarthome Zigbee Maxled RGBW Controller.
- Manufacturer: Paulmann Licht GmbH.
- Model identifier: RGBW light.
- Device type:
  - [x] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [ ] Other: 

## Problem
The controller is detected but is not working properly. Only the RGB part can be controlled.

## Screenshots

[image]
[image]

SwoopX, 2020-06-20:
> The controller is detected but is not working properly. Only the RGB part can be controlled.

Can you be a bit more specific on this part? Where?

ebaauw, 2020-06-20:
As it only exposes a single endpoint, it would seem the controller doesn’t allow independent control of the RGB and W channels.  It probably mixes in the W channel when using colours with low saturation, or when setting colour temperature.

cma-bw, 2020-06-20:
You're right. With increasing saturation, the RGB leds become brighter and the W leds become darker. Unfortunately the controller doesn‘t allow independent control like the FLS-PP.
Thanks for the hint.

closed 2020-06-22 by cma-bw