dresden-elektronik/deconz-rest-plugin #2972: [Device Support] Hive Active Plug SLP2

peterguy04, 2020-06-20:
#Device

- Product name: Hive Active Plug
- Manufacturer: Computime
- Model identifier: SLP2
- Device type:
  - [ ] Light
  - [ ] Lock
  - [ ] Remote
  - [ ] Sensor
  - [ ] Siren
  - [ ] Thermostat
  - [X] Other: Smart Plug

## Screenshots

 - Endpoint and Clusters of the node
[image]

  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]

[image]
[image]
[image]
[image]
[image]

Smanar, 2020-06-20:
Hello, just some questions:
- Do you know what can be the utiity at alarm cluster ?
- Your device have only consumption and not power ?
- The on/off is already orking if you are using Phoscon ?

peterguy04, 2020-06-20:
hey, 
1, sorry i arnt sure what you mean by this question
2, yes thats right, it can show me current watt usage for eg if my computer is on it will show 150w, if i shut down that will drop to 3w.
3, yes on/off works fine currently

Smanar, 2020-06-20:
Ha yes, You are right sorry, you have consumption AND power, I haven't see it at first time.

Can you try these modifications ? https://github.com/Smanar/deconz-rest-plugin/tree/SLP2

I have just disabled tension and freqency.

peterguy04, 2020-06-20:
sorry for the delay, i didnt know how to do it so tried to have a go.
 got that branch running in a ubuntu VM , got the socket connected on/off is working i noticed in smart metering - instantaneous demand now shows a value, i didnt notice this before . though i cant see any monitoring information in phoscon.

SwoopX, 2020-06-20:
Check the REST API and share what the `/sensors` output is. 1 sensor should have been created containing power and consumption. I guess that was it then already.

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

peterguy04, 2020-06-20:
ok so bare with me as this is all new to me but i think i was able to do what you asked, and get into the REST api however i am not seeing any new sensors other than the default one.
[image]

SwoopX, 2020-06-20:
Jep, the sensor is indeed missing, but you did alright. Might be related that @Smanar accidentially overlooked a detail in his commit. Once an update is available, give it another shot.

Smanar, 2020-06-21:
@peterguy04 use the same procedure than in first page but change the step 1 by

git clone --branch SLP2 https://github.com/Smanar/deconz-rest-plugin.git

And after file replacement, en reloading deconz, you need to make a new inclusion to have new sensor creations.

peterguy04, 2020-06-21:
i am probably doing something wrong here... i started again using ```git clone``` as you suggested and compiled it, i then repaired the plug, is that what you meant by make a new inclusion?  i am on the discord channel aswell if it would be easier for you to talk to me there (pete04g)

Smanar, 2020-06-21:
Ok, so after lot of tries, still un-succed.

The actual code is this one > https://github.com/dresden-elektronik/deconz-rest-plugin/compare/master...Smanar:SLP2

The fonction DeRestPluginPrivate::addSensorNode() is never trigger.

The last log : https://pastebin.com/cYHJJ1Rp

If someone have an idea ?

SwoopX, 2020-06-21:
I'll checkit again later. Right now, I wouldn't have done anything differently.

Smanar, 2020-06-21:
And we have tried to update the device API entry, with setting phoscon in permit join and press "read", nothing more.

Smanar, 2020-06-21:
Ok So if I m right
- deconz trigger DeRestPluginPrivate::nodeEvent()
- them in this fonction it select case deCONZ::NodeEvent::NodeAdded:
- them in this block it make  addLightNode(event.node()); and addSensorNode(event.node());

Why the fonction void DeRestPluginPrivate::addSensorNode() is not trigger (I have put a debug line just at start)

Do you think the cluster 0009 can change something ? It's the endpoint used in the UniqueID.

SwoopX, 2020-06-21:
> Do you think the cluster 0009 can change something ? It's the endpoint used in the UniqueID.

You mean the endpoint? Don't think so. Maybe it needs some extra whitelisting in `handleIndicationSearchSensors()` at the beginning? Had that with a Samsung smart plug...

Smanar, 2020-06-22:
Ok @SwoopX it's working.
The problem was from a skipped step in the compilation procedure.
so I can confirm this device is classic, and I m taking the 2 other hive device, I will make 1 PR for the 3 devices.

closed 2020-07-17 by peterguy04