dresden-elektronik/deconz-rest-plugin #2973: Can't readd Tradfri Dimmer after removing them

xxLeoxx93, 2020-06-21:
Hello,

two of my Tradfri dimmer dropped the connection over night and didn't respond anymore (a click didn't do anything), so I deleted them both in order to repair them via the delete function in Phoscon web.
Now neither of both can be readded, but they can still be paired to a tradfri gateway.
I reverted back from the latest beta to 2.05.77 before the issue happend.
Gateway firmware: 26580700
Device: Conbee 2
Host Rasberry Pi 4

  If the problem can't be reproduced and is sporadic, please provide some details
  on how often and when the issue happens.
-->

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

xxLeoxx93, 2020-06-24:
Update: More dimmer went offline - none of them I could readd anymore...

SwoopX, 2020-06-24:
Can you share a debug log of the join process with `--dbg-info=2`, please? This might eventually have something to do with a malformed manufacturer name, which should have been catched with .78. That seems to be a bug in the ikea firmware.

xxLeoxx93, 2020-06-25:
But then I could also just update to .78 to solve it right? I'll try to get the log this weekend.
Is there any chance that the deletion of the swich in phoscon wasn't successfull and now I can't see it there but it's still paired in the bacground?

cputoaster, 2020-06-27:
seems like I have the same problem, host: ubuntu 20.04 NUC, conbee 2, 2.05.77, but fw 264A0700, did not upgrade yet. No way to add new dimmers atm or make ones with bad batteries work again.

cputoaster, 2020-06-30:
I switched to 2.05.78 / 26580700 and it seems to work again but needs a few attempts (around 110 nodes)

cputoaster, 2020-07-14:
I seem to have this this coming up again, or maybe it's a different form. One node dropped off, maybe because of low batteries, but did not come back with new ones. So I tried to deleting it from Phoscon and re-adding it, and after a few attempts stopped trying, but saw this strange node with the name "0x5A9C" and (I think) the MAC address of the dimmer. Any ideas? Can I somehow remove this half-included node?

[image]

SwoopX, 2020-07-14:
Those are incorrectly paired devices. It typically helps if you enter sensor search and read the basic cluster. You currently can't remove them other than directly deleting all remnants in the database (which I wouldn't recommend).

closed 2020-08-14 by stale[bot]