dresden-elektronik/deconz-rest-plugin #2974: deCONZ has all my devices, but Phoscon App (and Homebridge) do not

mtressl, 2020-06-21:
To be honest, I am quite lost. After having a very stable system based on **2.05.77** for a while (even though without VNC) I did a Docker container `pull` for the latest version, and an `up -d` to restart. All was fine, Phoscon App gave me **2.05.78** as current version and all devices were still there, working perfectly fine with the Homebridge Hue plugin. I left that +/-24hrs until I decided to look into the RaspBee II firmware update again, since now the logs show there is an update available (I still have **26520700**, logs show **26580700**). So I did how described in the _readme.md_ - stop the container, and initiate the _FWflasher_. Since it (again) gave me the error of missing GLIBC version I stopped the process and brought my Docker container back with `up -d`, no problem.

**_Now it's the point I cannot wrap my head around._**
To check things, I access the Phoscon App. The Phoscon App lost my gateway name, and asks me for a new one and to set a password. _Ok, the shit hit the fan._ Phoscon App is empty, rooms and devices are gone. Bummer.
Desperate I try to access deCONZ via VNC (and did not expect it to work, since it did not work in **.77**). But hello, VNC works like a charm and I have a connection. AND all my devices are there in deCONZ. _WHAT???_
Ok, back to Phoscon App to check things, and now there is a _Group 420_ showing up, and the lights section has my two bulbs, one outlet and the range extender. Lights and outlet are in Group 420 (my former basement room), but all sensors, switches, and the roller blinds are missing (all passive battery devices).
My blinds still work, and I can move them up/down with the Ikea up/down switch (they were paired before). But all of the devices are not available in Homebridge, and not in HomeKit (ALL of them).

So a bit context, but I really do not see why this is happening:
- Pi 3B+ running Homebridge in docker, all latest versions
- the same Pi running deCONZ in docker
- Homebridge Config-Ui-X running on 192.168.1.xxx:8080
- Phoscon App running on 192.168.1.xxx:8282
- VNC (now again) running on 192.168.1.xxx:5900
- `docker ps` gives me two running containers (homebridge and deconz) as expected

Any ideas guys? Is it the missing firmware update that causes this? Why did Phoscon App lose everything, then partly reappearing, but nothing is exposed to anywhere?

Help is really appreciated. Let me know if further details (e.g. logs) might help.

mtressl, 2020-06-21:
Cross-post here, since I have no clue where the issue originates: https://github.com/marthoc/docker-deconz/issues/237

SwoopX, 2020-06-21:
What dors a REST API query for `/sensors` return? I'd assume your deconz database is intact...

mtressl, 2020-06-21:
Well the API query for `/sensors` returns the same I see in Phoscon App:

```
{
1: {
"config": {
"configured": true,
"on": true,
"sunriseoffset": 30,
"sunsetoffset": -30
},
"etag": "871da55a4e6dd38d7777c69b62ba71ad",
"manufacturername": "Philips",
"modelid": "PHDL00",
"name": "Daylight",
"state": {
"dark": false,
"daylight": true,
"lastupdated": "2020-06-21T04:15:04.336",
"status": 160,
"sunrise": "2020-06-21T03:26:50",
"sunset": "2020-06-21T19:24:21"
},
"swversion": "1.0",
"type": "Daylight",
"uniqueid": "00:21:2e:ff:ff:05:33:a9-01"
}
}
```

What do you mean by 
> I'd assume your deconz database is intact...
How to check?
The command docker logs deconz does not come up with errors as far as I can see, just quite regularly with:
`12:31:34:450 don't close database yet, keep open for 900 seconds`

SwoopX, 2020-06-21:
Ah, I misunderstood that then. The nodes you see in deconz GUI have only hex values as name?

The output lokks like a fresh deconz DB. I don't know where and how it is stored in docker nor if it is recoverable.

mtressl, 2020-06-21:
Yes, correct: devices in deconz GUI have only hex values (the passive battery ones). The active ones (lights, plugs, extender) came up with their generic names, and since they appear in Phoscon App I was able to rename them from there and now they are renamed in deconz GUI as well.

In the Phoscon App I see the following, but I am not sure if that helps or would make everything even worse (available through https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues):
[image]

SwoopX, 2020-06-21:
Really strange that just the sensors went away. Can you start deconz manually with debug output enabled? `--dbg-info=2 > debug`?

Would be interesting to see what happens on startup...

mtressl, 2020-06-21:
That would be a change in the `docker-compose.yml`, correct?

```
    devices:
      - /dev/ttyAMA0
    environment:
      - DECONZ_WEB_PORT=8282
      - DECONZ_WS_PORT=8443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
```

Change to `- DEBUG_INFO=2`?

SwoopX, 2020-06-21:
Jup

mtressl, 2020-06-21:
Ok, now what _could_ be interesting out of all that?

```
17:43:09:395 HTTP Server listen on address 0.0.0.0, port: 8282, root: /usr/share/deCONZ/webapp/
17:43:09:466 CTRL. 3.16.217:43:09:582 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:43:09:582 dev /dev/ttyAMA0
17:43:09:582 COM: /dev/ttyS0 :  (0x0000/0x0000)
17:43:09:582 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
17:43:09:817 parent process /bin/sh
17:43:09:817 gw run mode: docker
17:43:09:817 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
17:43:09:818 sd-card cid: 02544d534131364731ffffffff01191b
17:43:09:819 DB sqlite version 3.16.2
17:43:09:821 DB PRAGMA page_count: 30
17:43:09:821 DB PRAGMA page_size: 4096
17:43:09:821 DB PRAGMA freelist_count: 0
17:43:09:821 DB file size 122880 bytes, free pages 0
17:43:09:821 DB PRAGMA user_version: 6
17:43:09:821 DB cleanup
17:43:09:822 DB create temporary views
17:43:09:823 DB view [0] created
17:43:09:824 DB view [1] created
17:43:09:824 DB view [2] created
17:43:09:825 DB view [3] created
17:43:09:825 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
17:43:09:826 sql exec SELECT key FROM config2
17:43:09:826 sql exec SELECT key,value FROM config2
17:43:09:827 Load config UTC: 2020-06-20T18:58:13 from db.
17:43:09:827 Load config announceinterval: 10 from db.
17:43:09:827 Load config announceurl: http://dresden-light.appspot.com/discover from db.
17:43:09:827 Load config apiversion: 2.05.78 from db.
17:43:09:827 Load config bridgeid: 00212EFFFF0533A9 from db.
17:43:09:827 Load config datastoreversion: 93 from db.
17:43:09:827 Load config dhcp: true from db.
17:43:09:827 Load config discovery: true from db.
17:43:09:827 Load config factorynew: false from db.
17:43:09:827 Load config fwneedupdate: false from db.
17:43:09:827 Load config fwupdatestate: idle from db.
17:43:09:827 Load config fwversion: 0x26520700 from db.
17:43:09:827 Load config gateway: 127.0.0.1 from db.
17:43:09:827 Load config group0: 65520 from db.
17:43:09:827 Load config groupdelay: 50 from db.
17:43:09:827 Load config gwpassword: $1$8282jdkm$dsxave4Ld5bNO59zg6Dod/ from db.
17:43:09:827 Load config gwusername: delight from db.
17:43:09:827 Load config homebridge: not-managed from db.
17:43:09:827 Load config homebridge-pin:  from db.
17:43:09:827 Load config homebridgepin:  from db.
17:43:09:827 Load config homebridgeupdate: false from db.
17:43:09:827 Load config homebridgeupdateversion:  from db.
17:43:09:827 Load config homebridgeversion:  from db.
17:43:09:827 Load config ipaddress: 192.168.1.144 from db.
17:43:09:827 Load config linkbutton: false from db.
17:43:09:827 Load config localtime: 2020-06-20T20:58:13 from db.
17:43:09:827 Load config mac: 38:60:77:7c:53:18 from db.
17:43:09:827 Load config modelid: deCONZ from db.
17:43:09:827 Load config name: RaspBee II GW from db.
17:43:09:827 Load config netmask: 255.0.0.0 from db.
17:43:09:827 Load config networkopenduration: 60 from db.
17:43:09:827 Load config otauactive: false from db.
17:43:09:827 Load config otaustate: off from db.
17:43:09:827 Load config panid: 0 from db.
17:43:09:827 Load config permitjoin: 0 from db.
17:43:09:827 Load config permitjoinfull: 0 from db.
17:43:09:827 Load config port: 8282 from db.
17:43:09:827 Load config portalservices: false from db.
17:43:09:827 Load config proxyaddress: none from db.
17:43:09:827 Load config proxyport: 0 from db.
17:43:09:827 Load config rfconnect: 1 from db.
17:43:09:827 Load config rfconnected: false from db.
17:43:09:827 Load config rgbwdisplay: 1 from db.
[... 903 lines cut ...]
17:44:53:459 skip configure report for cluster: 0x0300 attr: 0x0004 of node 0x00158D0001FDF856 (wait reading or unsupported)
17:44:53:459 skip configure report for cluster: 0x0300 attr: 0x0008 of node 0x00158D0001FDF856 (seems to be active)
17:44:53:464 Websocket 192.168.1.144:33256 send message: {"attr":{"lastannounced":null,"lastseen":"2020-06-21T15:44:53Z","manufacturername":"MLI","modelid":"ZBT-ColorTemperature","name":"Keller ColorTemp Light","swversion":"2.0","type":"Color temperature light","uniqueid":"00:15:8d:00:01:fd:f8:56-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"00:15:8d:00:01:fd:f8:56-01"} (ret = 2122354688)
17:44:53:464 Websocket 192.168.1.217:62908 send message: {"attr":{"lastannounced":null,"lastseen":"2020-06-21T15:44:53Z","manufacturername":"MLI","modelid":"ZBT-ColorTemperature","name":"Keller ColorTemp Light","swversion":"2.0","type":"Color temperature light","uniqueid":"00:15:8d:00:01:fd:f8:56-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"00:15:8d:00:01:fd:f8:56-01"} (ret = 2122354688)
17:44:54:005 poll node 00:15:8d:00:01:fd:f8:56-01
17:44:54:005 Poll light node Keller ColorTemp Light
17:44:54:053 Poll APS request to 0x00158D0001FDF856 cluster: 0x0006 dropped, values are fresh enough
17:44:54:447 Websocket 192.168.1.144:33256 send message: {"attr":{"lastannounced":null,"lastseen":"2020-06-21T15:44:54Z","manufacturername":"Philips","modelid":"LWB004","name":"Keller Dimmable Light","swversion":"5.105.0.21169","type":"Dimmable light","uniqueid":"00:17:88:01:00:da:2f:99-0b"},"e":"changed","id":"5","r":"lights","t":"event","uniqueid":"00:17:88:01:00:da:2f:99-0b"} (ret = 2122354688)
17:44:54:447 Websocket 192.168.1.217:62908 send message: {"attr":{"lastannounced":null,"lastseen":"2020-06-21T15:44:54Z","manufacturername":"Philips","modelid":"LWB004","name":"Keller Dimmable Light","s [... cut ...]
17:44:56:337 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('5', 'normal', '00:17:88:01:00:da:2f:99-0b', 'Keller Dimmable Light', '65520,240', '11', 'LWB004', 'Philips', '5.105.0.21169', '{"attr/lastannounced":null,"attr/lastseen":"2020-06-21T15:44:56Z","attr/manufacturername":"Philips","attr/modelid":"LWB004","attr/name":"Keller Dimmable Light","attr/swversion":"5.105.0.21169","attr/type":"Dimmable light","attr/uniqueid":"00:17:88:01:00:da:2f:99-0b","state/alert":null,"state/bri":213,"state/on":false,"state/reachable":true}')
```

mtressl, 2020-06-21:
Well, in deCONZ via VNC all seems still there:
[image]

SwoopX, 2020-06-21:
Try doing a sensor search in Phoscon and push the physical button on a sensor. Alternatively, try to read the attributes from basic cluster.

mtressl, 2020-06-21:
Reading attributes from basic cluster did not work while in sensor search mode in Phoscon.
Will try to pair them tomorrow.

What really puzzles me is that the active devices, even they are available and active in deCONZ and Phoscon, do not show up through the Homebridge Hue plugin. Do you agree this is strange?

SwoopX, 2020-06-21:
There seems to be something wrong with your database, at least deconz wasn't able to find the sensors:
[image]

You might want to inspect the database file (`zll.db`) directly with sqlite. Make an `select * from sensors;` when you've opened it and also check the sensors table schema `.schema sensors`

Before you re-pair, I'd suggest open sensor search, choose one sensor and waken it (by e.g. weaving your hands, open close, warm it, etc.). Meybe they come back with that.

mtressl, 2020-07-18:
> There seems to be something wrong with your database, at least deconz wasn't able to find the sensors:
> 
> You might want to inspect the database file (`zll.db`) directly with sqlite. Make an `select * from sensors;` when you've opened it and also check the sensors table schema `.schema sensors`
> 
> Before you re-pair, I'd suggest open sensor search, choose one sensor and waken it (by e.g. weaving your hands, open close, warm it, etc.). Maybe they come back with that.

I ended up adding everything from scratch, starting with the bulbs and the repeater, finishing with the sensors. Since then everything seems stable and reliable. The whole installation even survived a power outage in between and came back normal. Have a lot more sensors now delivered I will add during the next couple of days.

MattWestb, 2020-07-18:
Thats sounds great.
And you was doing it in the right oder 2 !!
Don't forget making backup then its very easy getting corrupted database and killed SD-card (with or without power outage) and then you must redo all agen (Its the personality deCONZ).
If having moderate DB problems its possible ending the DB then deCONZ its offline but not recommended. 
I have one 6 month old DB and the table with groups was full of old deleted grupes (over 90 of 100 possible) and the device table wos having menny deleted nodes that was acting as ghosts in deCONZ GUI (I have more zigbee networks so much coming from the the other networks). 
deCONZ reusing the old information in a very bad hacked way on every program startup for making it faster and for handling upgrade from older version doing program upgrade (= no version handling of the DB). This making very much problem and the normal procedure (in larger DB systems) for program upgrade its having DB versions for the database, and by upgrade using that then upgrading and rebuilding the DB in the new version then the database structure its changed in the program (DB version its changed), and not doing some strange DB hacks every program startup that making problem of reusing old crap in the DB.
Hope you get all right with your new sensors and can enjoying your system fully.

closed 2020-08-15 by stale[bot]