dresden-elektronik/deconz-rest-plugin #2975: Restoring Gateway Configuration causes "segmentation fault" and deletes all lights

xxLeoxx93, 2020-06-21:
## Describe the bug
Hello,

since I've many issues like not working switches since I updated to the latest beta (and downgraded again) I tried to resore the zigbee configuration.
However this caused a segmentation fault and deleted all my lights.
Luckily after a some tries I still got the segmentation fault by my lights were back.

## Steps to reproduce the behavior

"Alt" + Click on the advanced tab in Phoscon Web app -> The restored the configuration from the last stable version I had installed.

## Expected behavior

I was hoping this would fix my not responding devices since I'll not readd ~70 devices by hand.

## Screenshots
[image]

## Environment

 - Host system: Rasberry Pi
 - Running method: Raspbian
 - Firmware version: 26580700
 - deCONZ version: 2.05.77 
 - Device: ConBee II
 - Do you use an USB extension cable: yes

SwoopX, 2020-06-21:
Can you please try to share a core dump?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian

xxLeoxx93, 2020-06-22:
> Can you please try to share a core dump?
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian

Do I need to replicate the issue afterwards? Because I'm very affraid of loosing all my lights again and not getting them back.

closed 2020-07-20 by stale[bot]